KiCad/kicad-footprints #866: Idea for IC sockets

leoheck, 2018-08-25:
I think that the IC sockets should come with the 3d model of the IC too. I always have to attach them manually. So I believe the process is easier to remove if needed.

poeschlr, 2018-08-29:
Wrong repo. 3d model stuff should be discussed in the packages3d repo.

closed 2018-08-29 by poeschlr

leoheck, 2018-08-30:
@poeschlr actually this issue is not about the 3d model stuff.  It is about referencing a 3d model inside of the footprint. It should be here because it is here that this reference is made.

For example, right now the footprint for DIP_24_SOCKET references only the 3d model of the a socket. I am just adding that it is better to add the model of the IC too. To make the compositiion

So, I am asking to give something like this:
[image]

Instead of this:
[image]

So, the 3d models list should be something like this: 
[image]

So, if someone needs to see the 3d without the IC, it can just disable the preview checkbox or deleting the reference instead of adding the reference itself which is harder to do.

poeschlr, 2018-08-30:
I would not use two references for that but generate the model such that it is a combination of both. You can not use the normal dip model anyways as it has the wrong z offset for this usecase so you do not save storage space by doing it the way you suggest.

leoheck, 2018-08-30:
True, saving storage is a good decision. But a new 3d model will use more space than a single z offset, isn't it?

poeschlr, 2018-08-30:
We do not allow offsets in the 3d model settings.