KiCad/kicad-footprints #932: DF12C(3.0)-50DS-0.5V(81)

jneiva08, 2018-09-10:
Hi,

I'm making footprint for [DF12C(3.0)-50DS-0.5V(81)](https://www.hirose.com/product/en/download_file/key_name/DF12C%283.0%29%2D50DS%2D0.5V%2881%29/category/Drawing%20(2D)/doc_file_id/38626/),  the strange thing is it didn't mention mention how should numbering the pad's (or i can't see it).
Can you give me your opinion?

thanks

evanshultz, 2018-09-10:
I don't see it either, including looking at the DF12 catalog page from Hirose's website. So weird. I can't recall not seeing pin 1 called out or showing the pin 1 mark in the plastic of the part.

It appears all the conductors are evenly spaced from the component center, though, so the numbering could be similar to the existing Hirose BM24 headers. (That being said, those two footprints aren't really consistent even amongst themselves...)

Perhaps in this case the user selects the numbering scheme. I personally like having rows of even and odd, for what that's worth.

jneiva08, 2018-09-12:
@evanshultz 
I did make all connectors of DF12C by script.
But if i give its original mame ex. DF12C(3.0)-50DS-0.5V(81) the kicad don't recognize because of "(" ")"
Can you advise the best option for the name?

Full name: Hirose_DF12_DF12C(3.0)-50DS-0.5V(81)_2x50_P0.50mm_Vertical.kicad_mod

Thanks

evanshultz, 2018-09-12:
I found https://www.hirose.com/product/en/products/part_no_conf/series_id/23/:
[image]

This character limitation is driven by KiCad itself and shown at http://kicad-pcb.org/libraries/klc/G1.1/. Parentheses are not allowed in footprint names even if you didn't run into this problem.

So, what to do?
1. I don't think readability is harmed if the first set of parentheses are removed, since there's a letter on one side and a hyphen after to separate the numbers.
2. And you can leave off the `(81)` at the end since that's the part packaging which we don't include in KiCad symbol or footprint names.

That would result in `DF12C3.0-50DS-0.5V`. Are you good with that? Do you feel that is clear enough?

jneiva08, 2018-09-12:
It makes sense.
I will modify the script to change the name and remove 80 because is discontinued. Then will open an PR

Thanks for the help.

closed 2018-09-12 by jneiva08