KiCad/kicad-footprints #955: How to name top-side slug footprints?

evanshultz, 2018-09-18:
@poeschlr @jkriege2 @Ratfink @Misca1234 @antoniovazquezblanco 
See https://github.com/KiCad/kicad-footprints/pull/947.

Check out the package from http://www.ti.com/lit/ds/symlink/tpa3126d2.pdf:
[image]

At https://github.com/KiCad/kicad-footprints/tree/master/Package_SO.pretty we have 4 slug up parts. two are manufacturer-specific footprint names and don't capture this. The other two include "EP" in the footprint name which doesn't quite seem right to me as the package has a slug but the footprint doesn't have an exposed pad. I think we should do better.

Do you guys have ideas on footprint naming if there's a top-side slug?

I suggest `HTSSOP-32_6.1x11.0mm_P0.65mm_SlugUp3.71x3.81mm` and just add the nominal slug size after `SlugUp` and omit the common `-1EP` which we can use only if the slug is on the bottom. Alternatively, we could do something like `EPUp3.71x3.81mm` or something by adding `Up` but that doesn't seem as clear to me.

LJJS, 2018-09-18:
My suggestion would be `HTSSOP-32-1EP_6.1x11.0mm_P0.65mm_EPUp3.71x3.81mm`.
The `-1EP` at the beginning is helpful to identify that the footprint has an exposed pad, and at the end of the name its defined which kind of EP is used.

Misca1234, 2018-09-19:
What does the 1 in '1EP' stands for ?

Another suggestion could be 'TopEP'

poeschlr, 2018-09-20:
I would name the footprints as if there is no EP and add a suffix. (After all the footprint does not come with an exposed pad)
The only reason for the suffix is such that we can have a separate 3d model for them. There is no difference on the footprint side to a package without such a pad. So add a suffix like `TopThermalPad[size x]x[size y]mm` (I know it is long but TopEP or TopPad might be confusing to users. It is also not really the same as the SlugUp thing we use elsewhere. That is for parts that are normally intended to be soldered to the pcb but are mounted the wrong way round. The slug is another name for the large "lead" of TO packages.)

DanSGiesbrecht, 2018-09-29:
@poeschlr @evanshultz @Misca1234 @Ratfink , revisiting this discussion with the hopes of finishing @LJJS 's PR's:
- https://github.com/KiCad/kicad-symbols/pull/889
- https://github.com/KiCad/kicad-footprints/pull/947

Personally, my thinking is that if a mechanical engineer is evaluating heatsink/case requirements, they won't rely on the KiCad-provided 3D model anyways, and would instead reference the datasheet for top-side exposed thermal pad dimensions. So, I would be content if the part didn't use a unique footprint.

That being said, I see the validity in creating a separate, unique 3D model. It does provide a reminder that there is an exposed area on the top of the part that needs to be taken into account.

I agree with a different name than "SlugUp" if that one's already being used.

Possible candidates that I can think of:
- TopThermalPad (as suggested by @poeschlr )
- TopHeatsink
- ThermalPkg

Maybe once we decide on a name here, I can open a KLC issue to clearly outline the use of "SlugUp" vs this use-case. Let me know what you all think!

evanshultz, 2018-10-01:
IMHO the options above are quite long. I suggested `SlugUp` for brevity but I understand having a small exposed thermal pad versus a top slug could differentiate an otherwise identical package.

Perhaps `TopPad[X dim]x[Y dim]mm`? It ties with the shortest suggestion above and don't imply a slug.

If you want to bring the mechanical engineer into it, and suggest they'll use different models, then I think we do need a unique footprint. How else will the mechanical engineer know a pair of SSOP-24 packages don't use the same model? By having a unique footprint we can embed that info into the design. And the footprint is associated with one and only one 3D model anyway.

Updating KLC would be good. It seems we've gotten rather lazy lately and let that doc being to rot.

DanSGiesbrecht, 2018-10-01:
I agree that shortness of the name is valuable. @poeschlr , you mentioned that `TopPad` or `TopEP` could be confusing, but if it's correctly documented in the KLC, I see no direct issue with that name.

antoniovazquezblanco, 2019-07-02:
This seems to have stalled. There are PRs depending on this. What would be the final veredict?

evanshultz, 2019-07-15:
I personally like `TopEP` and dimensions. I don't know why it would be confusing to anybody if they can already grapple with `EP` in the footprints names already, but if that's true then `TopPad` as I mentioned above.

Slugs are not only for TO-type packages. See every use of the "slug" at https://www.analog.com/media/en/training-seminars/tutorials/MT-093.pdf and https://www.st.com/content/ccc/resource/technical/document/technical_note/group0/89/b8/8f/9b/ed/7b/41/56/DM00454055/files/DM00454055.pdf/jcr:content/translations/en.DM00454055.pdf. It can mean the big TO lead, but that's not the exclusive use of the term.

evanshultz, 2019-07-22:
@poeschlr 
Can you please weigh in again? I would like to think we're almost finished with this.

I want to submit a footprint for http://www.ti.com/lit/ds/symlink/tpa3251.pdf, and there are other PRs dependent on this as Antonio mentioned above.

evanshultz, 2019-08-07:
@poeschlr ? Or anyone else? I would like to close this issue.

poeschlr, 2019-08-08:
First of all i do not quite get why names should be short. The 256 character limitation is long gone so why should we use shorthand here. I personally prefer human readable without need to check any outside source. (Otherwise we could just use the IPC naming convention.)

TopEP does not work in my opinion as it is not an exposed pad (exposed pad is a strange name anyways but well the industry chose this term.) Similarly with TopPad (pad describes something that is meant to be interfaced with. In the context of pcb design something that is soldered.)

To be honest from the suggestions made so far i would use TopHeatsink as it is the only thing that describes the function of that thing without additional knowledge requirements. And a metal thing on the top of a package can only have the purpose of being a heatsink. (It is not ideal but no other suggestion is quite as clear and i personally do not have a better alternative either.)

evanshultz, 2019-08-30:
If I have two valid choices and one is shorter, I'm inclined to use the shorter one. Nothing about shorthand, only choosing a shorter word or adopting abbreviations already in use in the library.

I need to clarify, so let me step through the above:
- You note the word "pad" indicates something to be interfaced with, such as the exposed copper on a PCB which solders to a pad.
- Footprints have pads while packages don't. So using "pad" (or a variant like EP) is referring to the footprint and not the package.
- But the word "pad" is very commonly used in the industry for exposed metal on a package, such as "thermal pad". That is also an interface which is soldered, but in this case there is a pad on the board and a pad on the package. I believe this is what you're saying above.
- `TopEP` or `TopPad` indicates a pad but this type of package doesn't have a pad for the central thermal interface on the top of the package, it has some kind of lead.
- As the footprint and 3D model are related it may be reasonable to include either footprint-specific or package-specific info in the footprint name.
- `TopHeatsink` doesn't describe the package. Nor the footprint. And the heatsink itself, which is optional, wouldn't be included in the footprint or 3D model.

I can't reconcile that suggestion. Maybe I misunderstood.

If the name of the footprint should include the footprint uniqueness and the 3D model uniqueness, then any of them seem to work because it would differentiate from a footprint/model without any metal on the top of the package.

While the package doesn't have a pad as I understand you defined it above, the industry uses the term term "pad" for a large non-outer lead. And "EP" is already uses to indicate a footprint with a large non-outer pad in the library so there's already a common understanding of the term in the library. And the footprint and 3D model need to be a unique combination. So I simply can't see why "TopEP" isn't reasonable. It builds on the terms used in the library, sticks with terms already used in the industry, meets the requirements for uniquely-defining a footprint:model combination, and is brief. That seems like a workable suggestion, though I'm open to something else that meets all the requirements (which I may have misunderstood or missed). On the contrary, "TopHeatsink" fails on many of those counts.

evanshultz, 2019-09-25:
@poeschlr 
Ping!

cpresser, 2019-10-12:
How about a `_ThermallyEnhanced` suffix?
[image]

chschlue, 2019-10-12:
I think _ThermallyEnhanced is misleading, HTSSOP is always thermally enhanced, just usually by having a bottom EP.

Two questions:
1. Top EP footprints are vanilla TSSOP, bottom EP ones are HTSSOP. Should these be called TSSOP or HTSSOP then? JEDEC MO-153 says HTSSOP has a thermal pad on the bottom while TI also calls these here HTSSOP, so following TI's naming would mean having loads of duplicate TSSOP footprints.
2. Should top EPs be present in symbols? If yes, how?

evanshultz, 2019-11-06:
Agreed. A lot of package types are 'thermally enhanced', and that isn't specific enough IMO.

1. The footprints for all are unique if we're to have unique 3D models. I don't think we can stick with a vendor's naming since it's specific to only one vendor. Does JEDEC have a package specification with a thermal pad on the top (I don't recall seeing one)?
2. I don't think so. If it attaches to a heatsink then the heatsink might have a pin on the schematic.

@poeschlr ?

chschlue, 2019-11-07:
1. I don't think so. The current lib is already quite messy when it comes to FPs with bottom EPs, so how about naming all standard packages without any exposed pad as usual, the ones with bottom EPs H<package>-1EP (as it is mostly the case currently) and regarding all the ones with top EPs as vendor specific until a standard comes up?
2. So they are completely ignored symbol- and footprint-wise if I understand you correctly? While this is an easy way to avoid FP filtering issues and such, I think it would be nice to be able to reflect that the top slug is connected to something via the heatsink in schematics somehow.

evanshultz, 2019-11-07:
1. I didn't even look at the TI datasheet, but it references JEDEC MO-153. I only see bottom-pad packages mentioned in that doc, though. From the footprint perspective, I'm fine with either nomenclature. Adding the leading `H` does clearly indicate they're 'thermally enhanced' which might be nice for filtering. As long as we have some way to differentiate standard, bottom pad, and top pad versions I'm good so we can have unique 3D models. And I still feel `_TopEP` works well.
2. That's my thought. I've added a heatsink symbol near the device that needs it and the heatsink symbol has the pin(s) required. I've also seen that in other schematics. If the top thermal pad has a pin it would get an electrical connection; is there a reason that net needs to be in the netlist?

chschlue, 2019-11-07:
1. MO-153 isn't really explicit about the thermal pad necessarily being on the bottom as far as I can see, it's just indicated by a bottom view showing a non-dashed pad. I assume that standardizing top EPs at least wasn't on anybody's mind there.
2. I guess you're right.

evanshultz, 2020-09-14:
@chschlue 
It's been a while and this is blocking at least one symbol PR. You're also in a different position now. Do you mind revisiting this conversation and seeing if there can be a decision to move forward?

chschlue, 2020-09-14:
I favor `_TopEPXxY` for brevity.

Tangent: we should push for better 3D model association options for v7. It's a pity we have to add redundant footprints again.

closed 2020-09-22 by evanshultz