KiCad/kicad-footprints #961: Resistor array footprint is missing

leoheck, 2018-09-21:
The R_Array symbol symbol does not have a footprint available

[image]

I only found this, but it is not the right one. 

[image]

It should be something like:
```
1 -- 2 
1 -- 3
1 -- 4 
1 -- 5
```

leoheck, 2018-09-21:
Example of datasheet
https://industrial.panasonic.com/cdbs/www-data/pdf/AOC0000/AOC0000C14.pdf

evanshultz, 2018-09-21:
First, the symbol you show above has only 5 logical pins, with one side of each resistor being bussed together. The Panasonic datasheet you linked has individual resistors so it wouldn't match this symbol anyway. For this datasheet, the footprint above may work (I didn't check dimensions) and you would need this symbol:
[image]

The R_Network03 symbol, for example, provides http://www.vishay.com/docs/31509/csc.pdf as a datasheet. This is a bussed resistor network (at least the CSCxxx01 version is) which matches the symbol. And it goes with this footprint:
[image]

https://industrial.panasonic.com/cdbs/www-data/pdf/AOC0000/AOC0000C12.pdf is a Panasonic SMT resistor network with a bussed configuration. Some of the variants could match the symbol above, but it's not a perfect match. And you need more than just 4 resistors to even try to make it work.

You're right that we could have a symbol that works well for SMD resistor networks as the bussed resistor symbol you showed above seems to be suited well for THT resistor networks.

Does all of that help? If you want to submit a PR to add more symbols we can certainly talk about the proper symbol name and get everything squared away so it's clear what could be added.

Also, I believe the pin numbers on this symbol should be shown, though, since this would make it clear which resistor is being connected to. Resistor swapping is common but awfully hard to do if one can't easily identify which resistor is which!

leoheck, 2018-09-22:
Thanks for the answer. I agree that the with all you said. Does it make sense to add more footprints considering I am using a symbol that doesn't have a SMD part? If yes, I can submit a PR. Do you have an initial proposal for the name of the footprint as you said? Maybe the core of the name can be something like `R_Parallel`

evanshultz, 2018-09-22:
The symbol you showed above may have a matching SMD component, but I just didn't find it in my quick search.

The R_Pack symbols I showed above already exist. Are those what you're mentioning? It's not clear to me what configuration you mean by "R_Parallel"; I've not seen internally paralleled network of resistors.

If you're talking about a bussed SMD resistor network, would you do a little research and see if there are common packages and pinouts? I noted a Panasonic datasheet above that has several configurations. If we do see common package and pinouts, adding a symbol would be a great idea. We would need to settle on a naming as you mentioned, but perhaps let's understand the scope first. :)

bobc, 2018-09-30:
After a bit of searching, I found only 1 part which is a bussed 4 resistor network. http://www.koaspeer.com/cnb/ and uses a non-standard footprint.

There are a lot more SMD packages containing 7+ bussed resistors, these appear to use standard footprint.

leoheck, 2018-12-30:
Hi guys, what should I do here? Should I close this issue?

herostrat, 2019-01-17:
Is you issue settled in a for you acceptable manner?

A brief look into the issue seems that it is resolved, at least if you did not find additional research and found a standard footprint for this odd/uneven number of pins.

If this is the case I guess you could close the issue :)