KiCad/kicad-footprints #973: ErrType(19) - Pad near Pad issue with Resistor_SMD:R_Array_Convex_4x0402

leoheck, 2018-09-26:
I am using `Resistor_SMD:R_Array_Convex_4x0402` and I always receive this `Pad near Pad` issue on PCBNew

[image]

I tried to reduce the track width to fix it. But nothing has happened. Now, it is 0.05 mm (1.97 mil) and I still have the Pad near Pad issue.

Any idea on how to fix it?

[image]

Shackmeister, 2018-09-26:
That is a matter of board/DRC setup.
Since its pad to pad and not track to pad, the tracks has nothing to do with it

closed 2018-09-26 by Shackmeister

evanshultz, 2018-09-26:
This repo is for discussing the libraries. One good place to post "using KiCad" questions is on the forum: https://forum.kicad.info/.

The constraint being violated is pad-pad spacing, so of course changing the track parameters won't help. This is controlled by the Clearance parameter in the Design Rules.

leoheck, 2018-09-26:
Thanks @evanshultz