KiCad/kicad-footprints #991: V5 Library browser issue

Sebastian-Kulik, 2018-10-03:
Hi,
I use KiCAD in my daily job. Mostly Nightly’s, but V5 too. V5 is installed in “KiCAD-V5” folder, Nigtlies in standard “KiCAD”. Im on Windows10x64(Pro).
Windows: 10-Pro, newest version (just all updated) x64bit
Workstation: TR1920X, 32G-RAM, Asus x399 ROG Mainboard, M.2 (System), SATA-SSD (Files)
V5 have issues in Footprint library browser - any ideas?
Best regards,
Sebastian
[V5.txt](https://github.com/KiCad/kicad-footprints/files/2441175/V5.txt)
[image]

herostrat, 2018-10-03:
Hello Sebastian,

this would be better reported here:
https://bugs.launchpad.net/kicad/+filebug

The code for KiCad is located on launchpad.
https://launchpad.net/kicad

This repository (and the sister repositories on github) are just for the libraries and their managment.

Regards

Sebastian-Kulik, 2018-10-03:
I fix the problem allready. It was easy.
in /appdata/Roaming/kicad  delete bugging "pcbnew", restart KiCAd- thats it!
After all: find the error compared new file with bugged file: it was wrong/error in window coordinates, but anyway, just let KiCAd re-build the config file, it works well.
Best regards,
Sebastian