KiCad/kicad-footprints #1000: Thermally-enhanced SO-8 footprint design (PowerPAK SO-8, LFPAK56, PQFN 5x6, etc.)

evanshultz, 2018-10-08:
See https://github.com/KiCad/kicad-symbols/pull/924#issuecomment-427667192 which surfaced a good question about footprint naming. I hadn't looked into it closely until this valid question made me do some Googling and then propose the below.

https://www.nxp.com/docs/en/product-selector-guide/75017590.pdf mentions many variations of a thermally-enhanced SOIC-8 package. There is no standard package which may be the reason for the proliferation of similar packages. NXP has come up with a single footprint that allows broad compatibility with package from many vendors.

http://www.onsemi.com/pub/Collateral/AND9137-D.PDF takes the above work and slightly modifies the stencil openings to give less voiding after reflow soldering.

Making a footprint combining the above seems like a nice addition to our library since we have several symbols that could use the footprint.

However, we already have a few footprints available and so this is my proposal to clean things up:
1. Remove `Package_SO:Vishay_PowerPAK_SO-8`.
1. Any Vishay symbols using that footprint use the other PowerPAK footprint. If the final, optional item below is done that symbol update could be done here too.
1. Add the new universal footprint. I don't have a good footprint name for it; any suggestions?
1. Any non-Vishay symbols using the PowerPAK footprints get updated to use the new universal footprint.
1. [Optional] Add a `Vishay_` prefix to the existing PowerPAK footprints. Since PowerPAK is a Vishay-specific name (both docs above show that and I don't find that term in other vendors' datasheets) it's already a manufacturer-specific name without adding "Vishay", but KLC indicates using the vendor name so this would do that. This would require a further footprint and footprint filter update to Vishay symbols.

At the PR mentioned above I gave instructions to move ahead. We won't be losing anything and I'll do any clean up required if the above is accepted or we come up with a different path. I'll also create the new footprint and do any symbol linking required.

evanshultz, 2018-10-22:
Anybody read this and support it?

herostrat, 2018-12-11:
I really like the idea and the PDF on onsemi was a very interesting read.

I guess because every vendor seems to use a different name the problem would be to find the correct one if one adds a new symbol for a project.

Alias would be nice, but are not supported until v6 afaik, so for overview and versatility I think the naming you propose is ok.

We could open another issue that has the goal to correct the lack of alias or at least track it.

flaviut, 2019-08-12:
Has anyone created the AND9137 footprint? If not, I'd be happy to contribute it.