KiCad/kicad-footprints #1003: Relay Sanyou SRD Form C - swapped pins

LubosD, 2018-10-09:
Hi,

the schematic symbol for the Sanyou SRD Form C relay has pins 1 and 3 indicated as connected in off state. The footprint, however, has pins 1 and 4 connected.

The footprint is correct. 

So either the drawing for the schematic symbol needs to be changed or the pins 3 and 4 must be swapped there.

See the pictures:
[image]
[image]

LubosD, 2018-10-09:
Here: http://www.sanyourelay.ca/public/products/pdf/SRD.pdf
And there's a Chinese compatible clone here: https://voron.ua/files/pdf/relay/JQC-3F(T73).pdf

Both have the same wiring diagram.

I have the Chinese clone in front of me and pin 1 is connected to the right pin (when viewed from the top - like in the footprint screenshot) in off state.

poeschlr, 2018-10-09:
Your observation is correct. I would suggest to change the symbol as the footprint now has the pads numbered in counter clockwise direction. (The numbering scheme would be strange if we would change the pad numbers instead of the symbol.)

evanshultz, 2018-10-12:
@LubosD 
Will you be submitting a PR to fix this? I don't want to leave an error in the library now that you've caught it.

Soldering-Stan, 2019-02-12:
Hi I've just encountered this problem too :-(  Unfortunately the board has come back from the fab house and I've had to remove a component and solder it elsewhere to fix the problem.
I downloaded the latest version of the library and found it still contains the bug.
It looks as if LubosD has not noticed your offer of a fix.
What's a PR and can I submit one if it will help get the footprint corrected ?

S

LubosD, 2019-02-12:
I wanted to submit a PR, but then I updated KiCad and pins suddenly seemed to be OK, which left me unsure if the bug is fixed or not or if I messed something in my install.

Soldering-Stan, 2019-02-13:
Maybe you were lucky :-)
I cloned this repo and md5 checksummed kicad-footprints/Relay_THT.pretty/Relay_SPDT_SANYOU_SRD_Series_Form_C.kicad_mod
and also the installed Relay_SPDT_SANYOU_SRD_Series_Form_C.kicad_mod and got 2b49a2c945763bc859f3b4d67337a0b4.  I also had an old library and got a different checksum.  When I did a diff on the old and new libraries, it does appear that pins 3 and four have been swapped.
47,48c47,48
<   (pad 4 thru_hole circle (at 14.15 6.05 90) (size 3 3) (drill 1.3) (layers *.Cu *.Mask))
<   (pad 3 thru_hole circle (at 14.2 -6 90) (size 3 3) (drill 1.3) (layers *.Cu *.Mask))
---
>   (pad 3 thru_hole circle (at 14.15 6.05 90) (size 3 3) (drill 1.3) (layers *.Cu *.Mask))
>   (pad 4 thru_hole circle (at 14.2 -6 90) (size 3 3) (drill 1.3) (layers *.Cu *.Mask))
(that's Old, New above).
I'll do some more digging and see if I can work out which library is correct.  It's all a bit confusing - I tried swapping the files and editing footprints in the application but they both looked wrong.  It's as if the footprints are cached and I'm picking up the wrong version.
Cetainly the footprint I used is wrong.  I just don't know if it's the footprint from this version of the library or from the previous version.

Soldering-Stan, 2019-02-13:
Hmm - something got confused - I didn't type that in like that - and I used no formatting characters :-)

poeschlr, 2019-02-13:
githubs formating is strange sometimes. I think i fixed it for you.)

poeschlr, 2019-02-13:
Oh and by the way footprints are cached in the pcb file. So if you change something in the lib (like updating it) you will only see the changes after telling kicad to update the footprints already placed. (This is on purpose to allow you to share your project or look at old archived pcbs without danger to the file no matter how much the library evolved in the meantime)

And you could still have the version 4 library if you updated to version 5 without taking care of it: https://forum.kicad.info/t/i-had-kicad-4-installed-previosly-now-i-updated-to-v5-now-i-have-some-problems-with-the-library-setup/11932

poeschlr, 2019-02-13:
I just checked this again. The issue still exists. So you might not really have a problem with the cache or the library setup.

I will open a pull request with the fix shortly.

Soldering-Stan, 2019-02-13:
Many thanks.  I'm an end user of Kicad and haven't dug into the inner workings so wasn't sure where the problem was.  I was going to create a couple of projects to try both versions of the library but you have saved me that - thanks :-)

poeschlr, 2019-02-13:
Fix is found in https://github.com/KiCad/kicad-symbols/pull/1527

Soldering-Stan, 2019-02-14:
Many thanks.  I'm an end user of Kicad and haven't dug into the inner workings so wasn't sure where the problem was.  I was going to create a couple of projects to try both versions of the library but you have saved me that - thanks :-)

I have checked the symbols page and commented.  It does look correct now.
The relays I have are made by SONGLE in case somebody is searching for a footprint for a SONGLE relay.

closed 2019-02-14 by evanshultz