KiCad/kicad-footprints #1009: PCB edge connector footprint datasheets

evanshultz, 2018-10-10:
As noted at https://github.com/KiCad/kicad-footprints/pull/996#issuecomment-428649751, none of our footprints have a datasheet attributing their design to a standard. That needs to be added and any dimensional changes to adhere to that standard made.

herostrat, 2018-12-11:
In addition to this I think it would be worthwhile to define how edge connector should be defined or if there are rules regarding them.

If you look into the PCIexpress one:
[image]

Imho the silkscreen is wrong on the button side and should be the same as in the others because it should not produce a silkscreen on the cuts on the final pcb
[image]

Also, is there a reason the pads are not rounded in that case?

I plan on doing the m.2 board connectors. 
Before I start I wanted to define how they should be done.

herostrat, 2018-12-11:
I agree, the edge cuts should be defined properly in the footprint.
But isn't this not possible until probably v6? I thought 5.1 would not include this option.

Otherwise I also think making templates for standard boardsizes is also very useful and sufficient for most cases (https://bugs.launchpad.net/kicad/+bug/1808011)

dlharmon, 2018-12-13:
Here's what I've been using since the 4.x days, seems to work fine with 4.05:

[image]

These are generated with this:
https://github.com/dlharmon/footgen/blob/master/examples/pcie.py

[pciex1fingers.zip](https://github.com/KiCad/kicad-footprints/files/2677846/pciex1fingers.zip)

poeschlr, 2018-12-13:
It is possible to define edge cuts. It however requires a bit of blasphemy (in other words a text editor)

In v4 it was even worse as opening a footprint with edge cuts lines meant they where automatically moved to the silk layer. Now they should at least stay on the edge cuts layer. (Maybe we can get the devs to allow a better workaround for 5.1 than using a text editor. Just help me out on this bug report: https://bugs.launchpad.net/kicad/+bug/1251393)