KiCad/kicad-footprints #1010: Unsure how to interpret datasheet

gcormier, 2018-10-11:
What are the two sections for? Is the outermost a pad, and the innermost a keep-out zone?

[image]
https://www.molex.com/molex/products/datasheet.jsp?part=active/0528921033_FFC_FPC_CONNECTORS.xml

evanshultz, 2018-10-11:
Yes, The boxes with cross hatching means "NO PATTERN" which is no copper in Molex parlance.

The boxes with parallel lines are the pads.

gcormier, 2018-10-11:
Would this just be representing on the footprint with a keepout in Dwgs.User? Such as this existing footprint:

[image]

evanshultz, 2018-10-11:
Due to KiCad being devoid of what any respectable ECAD program would consider a capable design constraint system, yes. See http://kicad-pcb.org/libraries/klc/F4.5/.

gcormier, 2018-10-11:
Dang. Yeah and drawing these lines, aligned, evenly spaced.... not much fun.

closed 2018-10-11 by gcormier