KiCad/kicad-footprints #1020: Right angle LED PTH footprints are backwards. Probably the SMT ones also.

cb88, 2018-10-16:
The cathode and anode are swapped on all the right angle LEDs relative to standard off the shelf LEDs.

Just open up any datasheet for a right angle through hole LED on digikey and you'll see with the LED facing you pin 1 (cathode) is on the right and 2 on the left. But all the footprints have this backwards.

This isn't as big a deal if you are using manually inserted right angle LED holder but if you buy a premade right angle LED.. it won't work.

The non right angle LED's look fine though.

evanshultz, 2018-10-16:
Thanks for reporting this potential issue.

Could you please provide at least one link to a datasheet? The parts you reference might not be the one I find. Do you find any datasheets where our footprints do work?

Then please confirm the footprints you refer to. Do you mean any one with `Horizontal` in the footprint name? For example `LED_D1.8mm_W1.8mm_H2.4mm_Horizontal_O1.27mm_Z8.2mm`, `LED_D3.0mm_Horizontal_O3.81mm_Z2.0mm`, and `LED_Rectangular_W5.0mm_H2.0mm_Horizontal_O3.81mm_Z5.0mm`?
[image]

cb88, 2018-10-18:
I'll provide at least 2 examples of each below (only these 2 of this type have any digikey stock anyway):
https://www.digikey.com/product-detail/en/vishay-semiconductor-opto-division/TLPR5600/TLPR5600-ND/4074661
[image]

https://www.digikey.com/product-detail/en/sunled/XSMG92D/1497-1359-ND/6615740
[image]

2nd Type:
https://www.digikey.com/product-detail/en/visual-communications-company-vcc/5300H5/L20015-ND/59969
[image]

https://www.digikey.com/product-detail/en/lumex-opto-components-inc/SSF-LXH104ID/67-1039-ND/270837
[image]

Not sure if type 3 exists? there are several similar designs but they all have board locks and varying pin arrangements.

jkriege2, 2018-10-20:
The footprints mentioned by @evanshultz are not meant for the types of LEDs you mentioned above. If you look at the 3D models this gets clear for case 3:
[image]
[image]

Maybe the closest to LEDs like your first two are side-emitters like this (there's only this one and some special ones in the Opto-Lib):
[image]
This footprint is for LEDs like this one http://cdn-reichelt.de/documents/datenblatt/A500/LED15MMGE_LED15MMGN%23KIN.pdf

For the horizontal LEDs in a plastic holder (cyour case 3; there are whole series of these e.g. from Mentor), there are no footprints at the moment ... if you have thime, you could add them? They would also be a good candidate for scripted footprints!

cb88, 2018-10-20:
Hand bent LEDs are irrelevant, you can orient them however you like... however the footprints are useless for the vast majority of available LEDs. From what I have seem 99% of available LEDs follow what I have said however if there are some that are in use that go the other way, perhaps the footprints should come in in either orientation, perhaps with a footprint name like LED_A1K2 LED_A2K1 etc... so it;s easy to tell. This really only matters for LEDs  that are at some angle or offset from the pins.

jkriege2, 2018-10-20:
Hand-bent might be irrelevant to you, but maybe not for others ... but that's not the point. The footprints are not MEANT for your case 3-LEDS ... footprints for these are simply missing! Even if they were oriented the other way, the silkscreen and F.Fab would not match your LEDs (also the pad to LED-distance might no correspond to the LEDs in datasheet case 3. Also the 3D model does not match you LEDs (e.g. because the black plastic is missing)

Therefore NAW/ADDITIONAL footprints & 3D models are required for the bent LEDs as in case 3. The same is true for the non-standard LED-packages in case 1&2. Fo all these mnufacturer-specific footprints would be best (the situation is comparable to e.g. all the screw terminals that we have in the lib, as they might differ slightly between manufacturers).

Would you be willing/able to contribute such footprints/3D models?

jkriege2, 2018-10-20:
PS: Most of the LED footprints are currently not manufacturer-specific, ... but they are also designed for the standard-LED-forms that are available from most manufacturers (3mm, 5mm, ...) If you have a look at the SMD LEDs, there are (in addition to the standard package 0805, 1205, ..., some PLCCs, ...) several manufacturer-specific footprints for LEDs with non-standard/manufacturer-specific shapes

jkriege2, 2018-10-20:
BTW: I tried to find some prebent LEDs (without the plastic), that would match the footprints we currently have ... but didn't find any that had a datasheet (only two from chinese shops without specifications&dimensions) ... are you aware of any?

evanshultz, 2018-10-23:
@cb88 
As @jkriege2 said, we don't have footprints for the parts to which you're referring. The best solution would be to add footprints for them and then I think all will be well. Since it seems like you're interested in these parts, are you up for adding the footprints?

evanshultz, 2019-04-09:
@cb88 
I gather the issue is that footprints don't exist for the parts you want to use. If correct footprints were submitted, then all would be well.

Can this be closed?