KiCad/kicad-footprints #1026: Please make smaller KiCad-Logo2 footprints!

ElectricRCAircraftGuy, 2018-10-17:
For really small boards, 6mm tall KiCad logos are still way too big! Currently the smallest "Designed with KiCad" logo is 6mm tall: KiCad-Logo2_6mm_SilkScreen.kicad_mod.

Please make 5, 4, and 3mm tall versions as well. Given the simplicity of the logo, these versions would still be easily readable, yet not too big for tiny boards that are only 1" x 1" or so. Thanks!

evanshultz, 2018-10-17:
Great idea. Please submit a PR to this repo.

evanshultz, 2018-10-17:
KiCad includes an image importer.

I do not know the origin of the logos used, but @SchrodingersGat may. Since they're being scaled so much a logo from the KiCad website may suffice. Please provide attribution of the image source in the PR when you submit it, unless somebody else chimes in with an answer first.

ElectricRCAircraftGuy, 2018-10-17:
`git log --diff-filter=A -- KiCad-Logo2_6mm_SilkScreen.kicad_mod` reveals the commit number and author. Here's the commit #: cf680ed33f6dcd0f8dc9180501ca4c08470fab68. https://github.com/search?q=cf680ed33f6dcd0f8dc9180501ca4c08470fab68&type=Commits. @SchrodingersGat appears to be the one who committed it, but I don't know if he made it or not.

ElectricRCAircraftGuy, 2018-10-17:
Here's the PR: https://github.com/KiCad/kicad-footprints/pull/50

diegoherranz, 2018-10-26:
Actually because the footprint repos were merged into a single one some time ago, the original commit for it was on the Symbols.pretty repo:

https://github.com/KiCad/Symbols.pretty/commit/71f14541191605f13fa68c5ea9db9434b092aeb9#diff-a65ab30cced17e6f360c67e89ba31913

There's a SVG link there.

herostrat, 2018-12-17:
https://github.com/KiCad/Symbols.pretty/commit/8f0c7423e40a22e9ffcae8913042583b99deaa16

codemonkeyrawks, 2019-01-02:
If we can I would like to see a 1mm logo, 2mm logo, 3mm logo, 4mm logo, 5mm logo being narrow. The board's I work with are typically 20mm at most and would look great if such logo could be added.

pointhi, 2019-01-04:
Please note the logo should be readable after manufacturing too :)

evanshultz, 2019-01-05:
Again, I believe the information is above to create new logos at other sizes. Then you may submit them to the library.

pointhi, 2019-01-07:
The "Designed with KiCad" was written inside PcbNew, exported as svg, added below the graphic and then imported as bitmap.

The easiest way to scale them down would be to write a script which applies a transformation matrix to every point of the polygon.

codemonkeyrawks, 2019-01-07:
@pointhi : can do. I will make a merge request when I am ready to have them be pushed in. I don't mind doing my own work as it's something I need to know how to do later on.

Thanks!

codemonkeyrawks, 2019-01-10:
@pointhi : thanks for the help: I have successfully updated a pull request here: https://github.com/KiCad/kicad-footprints/pull/1287. I am not sure yet if I will make a 4mm logo however 5 mm looks ok to use.

myfreescalewebpage, 2019-03-18:
@codemonkeyrawks I have just seen this issues, seems to be fixed with https://github.com/KiCad/kicad-footprints/pull/1287 and https://github.com/KiCad/kicad-footprints/pull/1312, right ? In this case I suggest we can close this issue. Joel

ElectricRCAircraftGuy, 2019-03-18:
I'm on my phone not a PC right now but it looks like that merge does not add 4 or 3 mm logos, does it? If it does not, then this issue is still open.

codemonkeyrawks, 2019-03-18:
@myfreescalewebpage : @ElectricRCAircraftGuy : I only created logo's that use: 5mm, 6mm, 8mm, 12mm, 20mm, 30mm, 40mm: (https://github.com/KiCad/kicad-footprints/tree/master/Symbol.pretty). As @pointhi suggested that anything real small may be more non-readable. 

I did not fix: 4mm or 3mm logos in the accepted patch however 5mm is there.

ElectricRCAircraftGuy, 2019-03-18:
Thanks. 
> As @pointhi suggested that anything real small may be more non-readable.
This issue is still open then, as there are tons of tiny boards people could really use the 4mm and 3mm logos on. Many boards are < 25x25mm in size, and the 3mm and 4mm logos would be perfect.

ElectricRCAircraftGuy, 2019-03-18:
@codemonkeyrawks , if you're willing to make the logos in 4mm and 3mm sizes too I'd be very appreciative, and so would many others. As far as readability is concerned, based on my experience with boards, **I have zero doubt that the main "KiCad" part would be very legible.** The readability of the "Designed With" part of the logo is certainly in question, but perhaps you could enlargen those words on the 4mm and 3mm logos to maximize readability of those 2 words.

myfreescalewebpage, 2019-03-18:
@ElectricRCAircraftGuy I have read above that 3 and 4mm logos will be very small to be readable and I agree with this. I suggest you should submit a PR with theses logos to check.

I'm not sure of the interest for logos on <25x25mm boards (where do you add the components if you put the logo ?)

Joel

ElectricRCAircraftGuy, 2019-03-19:
> I have read above that 3 and 4mm logos will be very small to be readable and I agree with this

Labels and other silkscreen text on PCBs is regularly as small as 32 mil, or 0.8mm. See JLCPCB's capabilities, for instance, here: 

> Min. Character Height | ≥32mil | Characters height less than 32mil will be unidentifiable.

I have personally printed characters as small as 0.7mm and had the boards manufactured through JLCPCB, and the text was still readable. 

I really can't understand where you are coming from. To claim that text that is 3mm tall is too small to read is rather mind-boggling to me, considering I have boards with readable text that is 0.7mm tall. Unless you are suggesting that all text on a PCB be constrained to 3mm tall or larger, I think that 3mm and 4mm logos are not only perfectly readable, but extremely easy to read. 

To further explain: font size 8.5 point on a PC is approximately 3mm tall. (8.5 point font x [~0.35mm/point](http://www.unitarium.com/font) = 2.975mm)

I'm really not understanding how you think 3mm is too small, or what evidence your argument is standing on. Small print at the bottom of banking documents is frequently as small as 5 or 6 point font, and again, any PCB designer regularly uses fonts down to 1mm or so.

codemonkeyrawks, 2019-03-19:
@ElectricRCAircraftGuy : I personally never said I was opposed to smaller logo changes nor am I a full member of KiCad - I  only help out where I can while I can. There is a great tutorial you can look at if your even wanting to try and tackle this. (https://www.youtube.com/watch?v=vaCVh2SAZY4) - PCB Series for KiCad.

I will try and look to see if I can create both 3mm and 4mm logo sets and will submit a PR request soon (this may take time - at least 3 weeks or so) depending on resizing and fixing up. I have Boise Code Camp this week first I must get some code up on.

myfreescalewebpage, 2019-03-19:
@ElectricRCAircraftGuy don't be so aggressive please, I'm just a member of Kicad who has review let says about 300 or 350 pull requests in the past 4 months, so I'm not a major contributor ;-)

My initial intention was to check if this issue is still valid because we have an important backlog and I regularly try to check at open issues to get news, and ask if closing is possible when they are terminated.

The last messages before my message above is the 10th of January and the pull request of @codemonkeyrawks has been merged in January too, so the issue start to be a little bit old without any news. You should simply respond as many other people do, "the original question is still open I would like to keep the issue opened", and if you explain the last status of the issue, what is missing or should be improved, with a simple comment and ideally a screenshot, it's a must to have.

Cheers,
Joel

codemonkeyrawks, 2019-07-01:
@electricRCaircraftguy sorry for no merge on the smaller logos yet. I have been busy with other things at the moment to really get onto fixing this item. I will submit a merge soon you just might have to wait some time for now.

codemonkeyrawks, 2020-09-28:
@ElectricRCAircraftGuy can you remind me again here what we discussed on what might need to change still. I have some time now I might be able to submit the changes as appropriate here.

Thank you.

myfreescalewebpage, 2020-09-28:
@diegoherranz all the messages above are just some unuseful comments, just refer to the first one of this thread: https://github.com/KiCad/kicad-footprints/issues/1026#issue-370925806

codemonkeyrawks, 2020-09-28:
I found looking thought the comments what was discussed and created (https://github.com/KiCad/kicad-footprints/issues/2499) for such issue. I don't see any point in keeping an old bug open if we addressed this PR.

myfreescalewebpage, 2020-09-28:
Why opening a new issue for something which is already discussed ? This just break the history of the exchange. Closing the new issue. If you want to submit a PR, link it with this issue please.

ElectricRCAircraftGuy, 2020-09-28:
> @ElectricRCAircraftGuy can you remind me again here what we discussed on what might need to change still. I have some time now I might be able to submit the changes as appropriate here.

Thanks, @codemonkeyrawks, I just checked the footprints in the repo again to be sure (https://github.com/KiCad/kicad-footprints/tree/master/Symbol.pretty), and the smallest "KiCad-Logo2" footprints are 5mm:

1. KiCad-Logo2_5mm_Copper.kicad_mod
1. KiCad-Logo2_5mm_SilkScreen.kicad_mod

In this ticket I am still requesting 4mm and 3mm versions of that logo. See [my first post](https://github.com/KiCad/kicad-footprints/issues/1026#issue-370925806), and the title of this ticket itself: "Please make smaller KiCad-Logo2 footprints! (5, 4, 3mm tall)". I'd like to leave this issue open until 4mm and 3mm versions are created at a bare minimum (update: 2mm too). _However, I think 2mm versions should be created too_, so I'm going to amend the title to state that. 

We should probably also make it standard practice to make logos down to 2mm height in general, and we should make a more simple logo which doesn't have multi-sized font in it, to go down to 1mm height, but both of these ideas should be new issues. 

And sorry I haven't had time to contribute these logos myself. I'd like to get back to contributing to KiCad when able.

ElectricRCAircraftGuy, 2020-09-28:
@myfreescalewebpage, I think we should, as a general practice, include the .svg files in the PR, as part of the library package, whenever we create logos or footprints with them too. Ex: this svg file should be committed, as well as all other image files we use to produce KiCad footprints: http://www2.futureware.at/~nickoe/kicad_logo_for_patch_monochrome.svg. How would you suggest we do this? 

Sample options:

1. For a KiCad footprint called `KiCad-Logo2_5mm_Copper.kicad_mod`, include the svg file as `KiCad-Logo2_5mm_Copper.svg`. Or, if a single svg covers many footprint files (I'm not sure on the mecanics of how to make a footprint from an svg yet), call it simply `KiCad-Logo2.svg`.
1. Make a new folder right at the repo root (https://github.com/KiCad/kicad-footprints) called `0_Graphics`. The 0 is just to force it to the top of the list to make it stand out. Inside this folder, you duplicate the folder structure of the footprints the graphic corresponds to. So, You'd put it here: `0_Graphics/Symbol.pretty/KiCad-Logo2.svg`.

I'm a fan of option 1 above to keep everything together in one place. We really need the graphics included whenever a footprint is made from them.

myfreescalewebpage, 2020-09-29:
Option 1 is my favorite too and you can submit like that if you make the new footprints. Do you have the original svg file to make the small footprint required ?