KiCad/kicad-footprints #1045: Fiducial inclusion in .POS file

thomasonw, 2018-10-23:
After  migrating to KiCAD v5, the board houses have been noting that Fiducials are not included in the .POS file.  I am able to point them to the .RPT file for the details, but is there a reason why they are not included in the .POS file as well?

I have noted that by manually editing the footprints and selecting 'Surface Mount' vs. 'Virtual' as the Placement Type will force the including of Fiducials in the .POS file.  Should be the default in the fiducial footprints?

(Perhaps also related --  [https://github.com/KiCad/kicad-symbols/pull/653](https://github.com/KiCad/kicad-symbols/pull/653) )

evanshultz, 2018-10-23:
So the behavior changed between v4 and v5? If so, that may not be a library issue. (A library change may be the solution, but perhaps not the source of the issue.)

I suggest you post at https://forum.kicad.info/ and/or https://launchpad.net/kicad for "how to" and possible bugs. This repo is only about the library, and we don't know your issue involves the library.

thomasonw, 2018-10-23:
Actually, I do not know.  I skipped over v4   

I do know V3 (2013-07-07) did indeed include added fiducials in the .POS files.

thomasonw, 2018-10-23:
@evanshultz  -- OK, comment back from launchpad:  _"I believe this is a library issue these are made as virtual footprints, but should be SMD if they should show up in the pos file."_

[https://bugs.launchpad.net/kicad/+bug/1799601](https://bugs.launchpad.net/kicad/+bug/1799601)

antoniovazquezblanco, 2018-11-22:
I made it work by editing every fiducial to be locked (in order to be able to update my board without them dissapearing) and surface mount (to make them available at the pos file) in my PCB.

[image]

evanshultz, 2018-11-22:
OK. So you didn't find another solution but to change their type to SM.

antoniovazquezblanco, 2018-11-22:
Yes, that is it.

antoniovazquezblanco, 2018-11-22:
Should we change this in the library?

evanshultz, 2018-11-22:
Yes, otherwise users will run into this issue again and again. I created https://github.com/KiCad/kicad-footprints/pull/1104 and would like to see it merged for 5.0.2.

closed 2018-11-29 by antoniovazquezblanco