KiCad/kicad-footprints #1064: Raspberry Pi Footprint

jneiva08, 2018-11-02:
Where can find raspberry pi / shield footprint?
If don't exist in witch library should be made?

evanshultz, 2018-11-02:
I think you're looking for https://github.com/KiCad/kicad-templates. These templates should be installed with KiCad by default and available when creating a new project from a template.

If that's not what you wanted please let me know.

jneiva08, 2018-11-03:
not quite.
The idea was to have footprint where can easily can change F.Fab to Edge.Cut to be easy make a shield or simply have idea the size.
Since kicad do not support Edge.Cut on footprints.

I have done the following footprint:
[image]

Where should be placed?

evanshultz, 2018-11-05:
So you want to make a shield or hat for a Raspberry Pi Zero and this is to capture the board outline so you can add the circuitry for your shield/hat? I feel like I'm not understanding you so sorry that I'm slow.

myfreescalewebpage, 2018-12-04:
@evanshultz : @jneiva08 wants to have a symbol + footprint to easily create RPi HATs. Meaning with the PCB dimensions included in the footprint itself to avoid future issues when creating the board.

I have exactly the same need on Mini-PCIe I have pull requested few days ago : https://github.com/KiCad/kicad-footprints/pull/1124.

I have created the Edge.Cuts lines with a text editor, not with the footprint editor, so when inserting the symbol in my schematic, then going to PCB editor, I get the correct dimensions of the board immediately and there is no possible mistake since the footprint is validated. That's really help.

There is no KLC about this subject (Edge.Cuts in the footprint) and it would be nice to unlock it in the footprint editor. To be discussed but probably all developers working with Arduino, RPi etc etc will be happy with that !

Cheers,
Joel

evanshultz, 2018-12-04:
I'm still not sure why a template won't work. As I mentioned above, I feel like I'm just not getting something and it's probably my issue. But I still am not sure where a template is falling short. Here is the interface for selecting a RPi Hat which does include a base schematic, board with outline, and circuitry pushed from the schematic. On top of this base, more parts can be added to the schematic to make all different kinds of Hats:
[image]

You will need to contact the devs about allowing Edge.Cuts layer on footprints. They are available at https://bugs.launchpad.net/kicad. There was discussion a while ago about it so perhaps you will have momentum on your side?

Module.pretty is for footprints of purchased modules. I understand this request to be for a canvas on which a project can be designed.

poeschlr, 2018-12-04:
Edge cuts in footprints are partially allowed in version 5 already (they are no longer moved to the silk layer when opening a board)

The edge cuts layer will always stay partly unsupported to communicate the fact that this is a dangerous layer to be used in footprints. (It will be grayed out but one will be able to move drawings to it using the properties menu.)

More details see:
https://bugs.launchpad.net/kicad/+bug/1251393

myfreescalewebpage, 2018-12-04:
> Edge cuts in footprints are partially allowed in version 5 already (they are no longer moved to the silk layer when opening a footprint.)

No, it is still modified on opening.

poeschlr, 2018-12-04:
Not in current nightly. But don't open the properties dialog for such an element. It will crash the footprint editor.

evanshultz, 2019-01-21:
@jneiva08 
I may be rather daft, but could you please help to elaborate on my questions at https://github.com/KiCad/kicad-footprints/issues/1064#issuecomment-444156117. I may be missing something but I'm not sure I understand and therefore could help to answer your question.

bobc, 2019-06-15:
> Module.pretty is for footprints of purchased modules.

The Raspbery Pi Zero is a module which can be purchased.

myfreescalewebpage, 2019-06-17:
There is a work in progress about this subject on the symbols, issue is at https://github.com/KiCad/kicad-symbols/issues/1772. Several PR has been submitted by @jstjst and I expect that he will be able to provide footprints too, but of course anyone can help on the subject!
Joel

jstjst, 2019-06-20:
I hope I will be able to get the remaining symbols pushed in near future. If someone would like to help with the footprints, I would appreciate it, so feel free.
Jonas

jstjst, 2019-12-22:
I (finally) started to work on the footprints. See issue #2016