KiCad/kicad-footprints #1067: Wrong SMD footprint for MeanWell IRM-02-XX

omogenot, 2018-11-05:
Hi,

There is a mistake in the footprint ACDC-Conv_SMMeanWell-IRM-02-XX. Looking at the data sheet (and the physical unit), there are only 2 x 7 pins (numbered up to 24) whereas the footprint in the library has 2 x 8 numbered up to 26.

evanshultz, 2018-11-05:
Thanks for posting this potential issue.

Including the datasheet and a screenshot would be helpful so someone can quickly verify.

The datasheet (https://www.meanwell.com/Upload/PDF/IRM-02/IRM-02-SPEC.PDF) shows:
[image]

Here is the footprint:
[image]

Yep, sure looks seriously messed up to me. And the IRM-03 SMD version appears to be equally hosed.

For completeness, here is the schematic symbol for the SMD version:
[image]

Fortunately, the schematic symbols looks good.

So it looks like pin 1 wasn't put in the right place, and the direction of pin increment was wrong, and the skipped pins were wrong, which all lead to only pin 13 being in the right place (and that just accidentally worked out). So this converter is totally unusable now. How very sad.

In addition to fixing the pin numbering, the fab and silk lines will need to be modified to correctly note pin 1's location.

It does not appear that these footprints are scripted, so I suppose they were handmade. Are you willing to submit fixed footprints?

Note: How did https://github.com/KiCad/kicad-footprints/issues/749 and https://github.com/KiCad/kicad-footprints/pull/750 get reviewed and merged but this serious bug remained??

omogenot, 2018-11-05:
Hi.
We're not looking for who made a mistake, I am the only one never making any :-) Don't worry.
I made the foot print mod, are you interested in it?

[ACDC-Conv_SMMeanWell-IRM-02-XX.kicad_mod.zip](https://github.com/KiCad/kicad-footprints/files/2550320/ACDC-Conv_SMMeanWell-IRM-02-XX.kicad_mod.zip)

antoniovazquezblanco, 2018-11-06:
Thanks, as soon as @evanshultz posts a PR I will take a look and merge it. Thanks!

evanshultz, 2018-11-06:
@antoniovazquezblanco 
I was assigning myself to the issue and didn't realize I was on the hook to make the PR. :) Hang on and I'll submit it in a few minutes.

@omogenot 
Thanks again for bringing up this issue and clarifying the datasheet's view.

evanshultz, 2018-11-07:
The fix is merged. Thanks again for reporting this issue!

closed 2018-11-07 by evanshultz