KiCad/kicad-footprints #1070: Capacitor footprint names

evanshultz, 2018-11-06:
http://kicad-pcb.org/libraries/klc/F3.3/ specifies something different than the existing footprints names for both our THT and SMD caps.

THT:
THT caps don't have height in the footprint name even though height is known (the 3D model has a height) so our footprints are really missing information right now. Putting the footprint selection in CvPcb for a generic part like this means the user doesn't have to choose the height until the schematic is finished. It is a bit of a burden, but otherwise aren't we making decisions (picking capacitor height) on behalf of the user which is worse? I think height and pitch should be required attributes for radial caps. Perhaps in trying to combine the different body styles this section of KLC isn't clear enough and it should be separated by body style?

SMD:
The footprints include the X and Y dimensions at the PCB (most often these are the same) and optionally the height. Our existing footprints use the cylindrical body diameter and height. Since these packages are normally named like our existing footprints and lead pitch doesn't vary much or at all, I think a KLC update is best. Also, KLC doesn't consider these caps can be nonpolar. Perhaps something like `C[P]_Elec_[Manufacturer]_[Diameter]x[Height]_[X]x[y]_[Modifiers]_[Options]` with `Manufacturer`, `X`, `Y`, `Modifiers`, and `Options` being optional?

Relevant PRs:
- https://github.com/KiCad/kicad-footprints/pull/1019
- https://github.com/KiCad/kicad-footprints/pull/1030
- https://github.com/KiCad/kicad-footprints/pull/1032

seppestas, 2019-08-28:
I also find it pretty disturbing the library and KLC do not match on this. The KLC can also be seen as a form of documentation for the library, meaning right now the documentation does not match what is there, potentially causing a lot of confusion and mistakes.

IMHO it is better to have no documentation / KLC than having incorrect documentation.

poeschlr, 2019-08-28:
I would not use [diameter]x[height] as this could be confused with our normal [x]x[y] syntax. The proper fix would be to use D[diameter]mm_H[height]mm.

We could document the old style in the KLC. But with a note that no new contributions are allowed to be made using it and that it will be declared obsolete with v6 (could be a "can not release v6 till fixed" thing).

Might however be best to have a separate section that handles such known violations as i suspect there will be more.

--

Yes the KLC would ideally document how everything is right now. But in reallity it defines how new contributions must look like. (Some legacy stuff simply was forgotten to fix. The electrolytic caps seem to be such a legacy thingy.)

lorem-ipsum, 2019-10-11:
I wanted to add some more SMD caps to the library (from (Panasonic)[https://industrial.panasonic.com/cdbs/www-data/pdf/RDE0000/ABA0000C1181.pdf]), but stumbled into this issue along the way. Mainly I am not sure whether/how manufacturers should be added.

Current case are slight differences in the lead shapes of Nichicon and Panasonic, e.g. for 6.3x5.8 packages. I could add the manufacturer name in these cases, but I think it would be more consistent to have them on every footprint (similar to the `Capacitor_Tantalum_SMD` library). If all have them it might be better to have them towards the end, so that they are still sorted by size first.

I also realised that the definition list in the footprint generator are often based on the pad sizes of the manufacturers instead of IPC calculations. I would be up to update that list and change the generator to produce "correct" names to update all existing CP_Elecs for v6.

poeschlr, 2019-10-11:
The tantal case code was left in for legacy reasons. In reality the EIA code fully specifies the capacitor.

poeschlr, 2019-10-11:
Here therefore my suggestion
`C[P]_Elec_[Manufacturer]_[part number]_D[Diameter]mm_H[Height]mm_[Modifiers]_[Options]`

Manufacturer and part number belong together and are both optional (there to have a way to represent strange highly specialized manufacturer specific parts.)

lorem-ipsum, 2019-10-11:
This sounds good, but could generate issues as the lead sizes are not captured at all (they rarely are) and the size of the ground pad is not captured. Might be that this is not an issue.

I will try to find definitions for all existing footprints, update them in the generator with tolerances and see if any issues show up.

evanshultz, 2019-10-11:
> I also realised that the definition list in the footprint generator are often based on the pad sizes of the manufacturers instead of IPC calculations.

I re-wrote the generator to support IPC rules but left the existing footprints alone if all pad info is provided instead of lead info. The non-polarized footprints use IPC rules.

@poeschlr 
Does you proposal for footprint name above take into account SMD and THT parts? SMD e-caps are not usually named this way. I have found the diameter and base X being common, sometimes also including the base Y dimension as well.

lorem-ipsum, 2019-10-11:
> I re-wrote the generator to support IPC rules but left the existing footprints alone if all pad info is provided instead of lead info. The non-polarized footprints use IPC rules.
I've seen that and used it for those I wanted to add. I am currently motivated to change the existing ones to the new generator as well to have a consistent base for the future. But I just found out that for some manufacturers it is difficult to specify the dimensions correctly as they only state maxima in some dimensions (e.g. outside lead length for [these](https://www.vishay.com/docs/28395/150crz.pdf)). Not quite sure how to handle this correctly.

poeschlr, 2019-10-12:
If the base size is needed as well then maybe (not sure if diameter should be before the bounding box size or after it)
`C[P]_Elec_[Manufacturer]_[part number]_[size x]x[size y]x[Height]mm_D[Diameter]mm_[Modifiers]_[Options]` 

or (not really KLC compliant)
`C[P]_Elec_[Manufacturer]_[part number]_[size x]x[size y]mm_D[Diameter]mm_H[Height]mm_[Modifiers]_[Options]`

lorem-ipsum, 2019-10-14:
> The base size and diameter are typically the same size so i wonder if this is not redundant information.

This is not true. It depends on the manufacturer, e.g. Panasonic usually has slightly larger bases ([example](https://industrial.panasonic.com/cdbs/www-data/pdf/RDE0000/ABA0000C718.pdf)).

In general I would be in favour of using manufacturer specific namings, as they have slight differences in lead dimensions. While this is not always the case it is sometimes. Having manufacturer specific naming for everything would make it consistent for all (using the package/size code as `part number`). Otherwise there will be one "standard" footprint for one size which was the first to show up and all others will be manufacturer specific. The only alternative I see is to use the maximum dimensions and hoping that they do not contradict anywhere.

I also think there should be a consistent naming (as `Modifier`) for vibration-proof mounts. Currently there are already some (e.g. `CP_Elec_18x17.5` from [150 CRZ of Vishay](https://www.vishay.com/docs/28395/150crz.pdf)). I am not quite sure how to name them. There are two variants I encountered so far. The dominant variant are supportive terminals next to the standard leads. The second are two extra pads on the side (both show up in the 150 CRZ series, the first in the Panasonic series linked above). For the former some manufacturers use one large rectangle (e.g. Vishay), while others use smaller pads for the supports (e.g. Panasonic). I was thinking of something like `_SupportiveTerminals` and `_SideTerminals` as suffix.

poeschlr, 2019-10-14:
Well i would say we provide generic footprints that fit most applications and let the user handle creating footprints for their specific part if they have higher standards. Similar to how we do it for the normal chip packages (0805 for example is made to fit even the worst tolerances. This is why the clearance between the pads is surprisingly small for it)

Otherwise we will end up with way more than 10000 footprints just for electrolytics and i guess an order of magnitude more for standard passives.

lorem-ipsum, 2019-10-15:
Ok. I will just add more components and see how it is going. Currently I think one can get rid of the base size as the maximum value seems to be the same most of the time, just the nominal value and tolerances fluctuate a bit.

Another question (maybe not so much about the naming): Currently the manufacturer and series are included in the description of the cap. Should we stick to it?

antoniovazquezblanco, 2019-10-15:
Hello @lorem-ipsum 

Beware of https://github.com/pointhi/kicad-footprint-generator/pull/396

I was trying to regenerate all the caps under the new name standard but you seem to be doing a more thorough job. Please, feel free to use my changes or request changes in that PR as I am willing to implement them ;)

Thanks!

poeschlr, 2019-10-15:
All dimensions in the footprint name should be nominal sizes. Sometimes manufacturers do not give nominal (especially for the height) so one is forced to use maximum.