KiCad/kicad-footprints #1088: Stencil configuration

jneiva08, 2018-11-16:
On RF_Module:ublox_SAM-M8Q, in the pads for stencil have Copper:F.CU and F.Past selected.
In Texas_VQFN-RHL-20 have pads for stencil are Cooper:None and F.Paste selected.

Witch are correct?

evanshultz, 2018-11-16:
The uBlox one is wrong. Paste pads should not have copper turned on.

In reality it won't matter here since the paste pads are totally inside the copper pads (and I think that should always be the case), but it's still against our paradigm. The check scripts are weak in helping to catch paste pads and so it's unfortunately easy to miss this during a review.

jneiva08, 2018-11-16:
I'm doing a new footprint for ublox 4G modem.
I never remember how need to make F.Paste, and was comparing the footprints and found it.
Later I will fix it.

jneiva08, 2018-11-16:
Have changed in the following PR [#1093](https://github.com/KiCad/kicad-footprints/pull/1093)

evanshultz, 2018-11-16:
That PR is merged. Thanks again! If this issue is answered, please close it.

closed 2018-11-16 by jneiva08