KiCad/kicad-footprints #1095: KLC vague on dimensions to use w.r.t tolerances

johnbeard, 2018-11-18:
The KLC documentations is vague on which dimensions to use. As I understand it, the implicit rules are:

* F.Fab: the shape is the *nominal* body outline of the part from the datasheet
* F.SilkS: The silk is just outside the *nominal* shape (so outset from the F.Fab by half the line width plus, presumably, "a bit"). The "bit" is also not well-defined, if the pads are not the outermost elements.
* F.CrtYard: 0.25mm (or more for certain types) from the outermost of the pads or body.

This could cause issues when a part has wide tolerances of the body. For example, imagine a (THT, so no pads outside the body) part that is 10x10mm +/- 1mm. If you draw the silk just outside the nominal, and a courtyard 10.5mm wide, if you get a part that is in-spec at 10.9x10.9mm, you will find it not only covers the silk but also spills out of the courtyard and might not fit when assembled.

I think the KLC should spell out in detail which dimensions one should use for each of Fab, Silk and CrtYd layers, and whether or not to account for tolerances specified in the datasheet.

evanshultz, 2018-11-19:
You are correct that KLC could use some improvement here. Some of these have been discussed already but I'm too lazy to dig up the relevant conversations.

The only comment to make is that the standard offset of silk lines from fab lines used in most of the librarian-developed (or -inspired) scripts use 0.11mm. This means the edges of the fab and silk lines touch. KLC specifies that silk should not be under SMT parts (and your comments about excessive tolerance may also have an issue with that) and 0.11mm offset is the minimum value that meets this rule.

Do you have a part in mind with dimensions and tolerance as described above, or are you only hypothetically postulating issues and pointing out good areas where KLC can be improved?

johnbeard, 2018-11-20:
A little bit of both.

I was looking at several parts like this:

[image]

The case only as a maximum size defined, and the position of that case WRT pin 1 is defined only in terms of maximum and minimum offsets from pin 1. Thus, in the x-axis only:

* The case edge on the left:
** Leftmost is position is as far left as possible: -2.62mm
** Rightmost is position is as far left as possible: -1.92mm
* On the right:
** Rightmost: when the case is at maximum and as far right as possible: -1.92 + 21.87 = 19.95mm
** Leftmost: not defined, as the case dimension doesn't have a specified "nominal" or "minimum" size, but in theory it would be -2.62 + case_x_min.

There is no explicit "nominal" for the position of the case, but you could average max and min and synthesise one. In this case, you get the left edge at -2.27. So the courtyard is at -2.27 - 0.25 = 2.52, and the silk line is centred on -2.27 - 0.11 = -2.38. If the case were at the leftmost position (-2.62), it's outside the defined courtyard by 0.1mm and covering the silk completely.

If you place two of these side-by-side courtyard-to-courtyard, and one was offset left and one right, you *could* get an overlap of 0.2mm of the physical parts. So the courtyard has failed its job, as you really wanted *at least* 0.5mm space, you got -0.2mm and you can't even get the parts into the PCB.

The KLC doesn't say the silk for THT *must* be visible (as you can reasonably put silk under a part), but IMO, an *outline* like this should be. Otherwise when assembling, it would look like something is wrongly placed.

Using the synthetic average placement size and the given case size max sizes for the Fab layer seems fine to me, it's a plausible graphic representation of the part.

To ensure this:

* Fab box from -2.27 to -2.27 to  21.87 = 19.60
* Silk from -2.62 - 0.11 = -2.73 to -1.92 + 21.87 + 0.11 = 20.06
* Courtyard from -2.62 - 0.25 = -2.87 to -1.92 + 21.87 + 0.25 = 20.20

But you would certainly get "empty" spaces inside the silk and courtyard when assembled, but you are guaranteed that the part will never cover the outline silk or go into the 0.25mm courtyard DMZ.

evanshultz, 2018-11-20:
I see. Thanks for the info.

Yeah, this drawing is under-constrained compared with most drawing I can recall. In this case, since MAX is the parameter they specify, I'd also be inclined to go generous and try and accommodate all possible individual parts that could be placed on a PCB. I'm not sure if there's any other way to do it, and with the official KiCad library being generalized I think having unused space on a board is vastly preferable to parts crashing.

IMO, this is a unique case and it may not be worth documenting in detail. The first few paragraphs at http://kicad-pcb.org/libraries/klc/ try to catch weird cases that only human intervention on a case-by-case basis can handle, and adding endless clauses for rare cases clutters up KLC for 99% of the common cases.

That all being said, you bring up many good points about valuable updates to KLC. I've added a few notes to my list of KLC updates which I'll include in a PR once I wrap up a couple other things. When the time comes I'll look into this more but I don't have any specific to contribute now. If you want to add or edit the existing KLC, certainly you're welcome. We can discuss more here or you can go right to a PR.

johnbeard, 2018-11-20:
I understand the KLC can't exhaustively go through corner cases.

I was thinking perhaps adding some fairly general guidelines, based on desired outcome. E.g.

* Fab layer outline are generally based on the nominal body size.
    * in this case there is no specified nominal, but a synthetic one from the average will do.
* Silk outlines should be visible for any variation of the part geometry that is within specified limits.
    * Perhaps a note of the de facto 0.11mm outset thing, even just as a note, rather than a rule? 
* Courtyard outlines should maintain the relevant clearance (usually 0.25mm) regardless of part variation.

johnbeard, 2018-11-21:
I've made a provisional PR at kicad-website, but I am totally fine with having it closed and addressed differently if that's preferred.

https://github.com/KiCad/kicad-website/pull/355

poeschlr, 2018-12-04:
The requirement for nominal dimensions comes from industry standards. The reason for this is that maximum tolerances are normally never really reached (very low probability)
It is also hard to determine what maximum means. Especially for long chains of dependent dimensions.

johnbeard, 2018-12-12:
@poeschlr I'm OK with it being nominal dimensions, too, as long as it's documented. I suppose the risk of collision is pretty low for normal population density anyway.

Whatever is used, I'd still like to see these norms spelled out in the KLC for each of Fab, Silk and Courtyard.

poeschlr, 2018-12-12:
Just to be clear: If only max is given then we can only use max. But if we have the choice then nominal should be used.

johnbeard, 2018-12-13:
@poeschlr, so, in the "normal" case:

* Fab is lines centred on nominal
* Silk is nominal, outset by a small amount - so that silk and fab lines just touch. This is generally therefore 0.11mm.
* Courtyard is nominal, outset by 0.25mm (or more for certain parts)

"Nominal" means that first of:

* The nominal size if given in the datasheet (eg 42mm +0.2mm - 0.3mm --> 42mm)
* Else the average if just the max and min are given (40mm to 41mm --> 40.5mm)
* Else the max if that's all there is

poeschlr, 2018-12-13:
About right. For silk we really do not have strict rules other than the must be outside the nominal part and have a clearance to pads (0.2mm for large parts. line with for smaller ones.)

The courtyard has a nominal clearance depending on device class. For most devices it is 0.25mm. (Details see KLC)
Ideally you would make a "temporary" polygon around the body outline and every pad with that clearance applied. Then union all these "temporary" polygons to get the correct outline. It is allowed to simplify this. Even down to having a single rectangle that only takes the bounding rectangle of the footprint as its basis. (The more detailed the courtyard is the fewer false positives will occur in the DRC check. The tradeoff is the increased work required to design and check it. In addition the calculation effort for DRC will also increase with very complex courtyard outlines.)