KiCad/kicad-footprints #1102: Push Button 12 mm has very thin holes

leoheck, 2018-11-21:
My 12mm push button can't fit on top of  `Button_Switch_THT:SW_PUSH-12mm` footprint. 

The holes should be increased a litle bit. 
It can be increased at least with the holes size used in `Button_Switch_THT:SW_PUSH_6mm`.
Which means changing the hole diameter from `0.8128` to `1.1`.

[image]

[image]

poeschlr, 2018-12-04:
You are right the hole sizes are too small. The datasheet linked to this footprint would require a hole size of 1.3mm. Right now it has 0.8128.

It looks awfully like the pad sizes and hole sizes have been convered from imperial measurements. So this fp might have been unchanged in the lib for quite some time. I would assume it is that way since before v4.

leoheck, 2018-12-04:
I discovered that one day after to place my PCBs order. I just replaced the small push button with the bigger one then I placed the order. I didn't have time to verify it at that time. 

Its ok, the workaround should not be complicated.

I will be glad to see this fixed on the next library update.

leoheck, 2018-12-30:
@poeschlr do you know if there is any way to verify the whole footprint library looking for issues like this?

crazyquark, 2019-06-28:
I was hit by this too(sadly discovered after making the PCB).

leoheck, 2019-06-28:
OMG, does this still have problems?

Who responds for this repo? Should we fix it and send a PR? How it works to
accelerate the bug fixes?

evanshultz, 2019-06-28:
@leoheck 
How would you verify the library for this kind of issue? The datasheet, and in some cases the physical part, are required. Do you have a proposal for how this could be automated?

Unfortunately, some parts of the library are crufty and this was a bad footprint. Or perhaps it worked for somebody but not for your switch. It's not well-specified.

New contributions are more rigorously reviewed and checked but reviewing and updating all old parts is done slowly if at all. It's sad when we find problems like this, but it's great you had the initiative to mention it here. Following up with a PR is the best way to make it happen. This is a community library with a few volunteer librarians and your support for new contributions and fixes are very important. Someone has to make a personal investment in time actually make the fix, not just mention it, to get the fix in the library.

@crazyquark 
Thanks for making the PR to correct this. Sorry about the bad boards. :(

leoheck, 2019-06-29:
@evanshultz thanks for answering.

I was not talking about automation, I was impressed that it wasn't fixed since ` Nov 21, 2018` when I reported it. I agree we don't have a way to check things like this automatically without human intervention. But it was an issue report. The fix is easy, and a bigger hole, in this case, is better than a small hole.

But, it is ok lets move on and @crazyquark thank you too for the PR.

> Following up with a PR is the best way to make it happen.

Cool, next time, I will make the PR and the issue report at the same time, so we can work together.

evanshultz, 2019-06-30:
Merged https://github.com/KiCad/kicad-footprints/pull/1667.

closed 2019-06-30 by evanshultz