KiCad/kicad-footprints #1105: outer pads of SMD resistor arrays are not placed correctly

tolot27, 2018-11-23:
If you look at #973, 
[image]
as well as [cra04p.pdf](http://www.vishay.com/docs/31048/cra04p.pdf), you can see that the outer pads are not placed correctly.

tolot27, 2018-11-23:
I cannot create a PR until I know what happened there: https://github.com/pointhi/kicad-footprint-generator/issues/220#issuecomment-441329659

evanshultz, 2018-11-24:
I'm not sure why the linked question above, about Phoenix terminal blocks, has any relevance here...

Also, a misunderstanding of the design rules does not mean the pad placement is wrong...

There's no enough information above to chase down a potential issue. If the footprint above is not meant for the CRA04P part, then it may not match. Perhaps the 3D model is wrong? Here is what would be helpful: To what footprint(s) are you referring? It appears to be one of the `R_Array_Convex_4x` ones (though not the 1206 version) but there's no way for anyone but you to know. And none of those footprints reference the CRA04P datasheet.

tolot27, 2018-11-25:
> Also, a misunderstanding of the design rules does not mean the pad placement is wrong...
So I misunderstood something?

Sorry for the inconvenience. Likely, it was to late to create the ticket because it does not fulfill my own requirements.

I have used the Resistor_SMD:R_Array_Convex_4x0402 footprint which referes to [mnr_g.pdf](http://rohmfs.rohm.com/en/products/databook/datasheet/passive/resistor/chip_resistor_network/mnr_g.pdf) and not cra04.pdf. I'll check further, what is wrong.

evanshultz, 2018-11-26:
The footprint looks like it's correct for MNR04 to me, so the 3D model may not be built to the same document.

Here are a few from our current library (convex 4x0402, convex 2x0402, convex 4x0603, concave 4x0402):
[image]

Thanks for digging up any more information.

Looking online, it appears that there are actual a number of different packages for "0402", isolated, quad, convex, resistor arrays. Perhaps our footprints names should be more specific since not all parts that would seem to fit a `R_Array_Convex_4x0402` footprint are actually compatible.

poeschlr, 2018-12-05:
You are correct about the fact that we do not yet have a script for these type of parts.