KiCad/kicad-footprints #1113: Footprint 12BH611-GR: wrong pins

frarees, 2018-11-26:
I used 12BH611-GR as a footprint for a Device/Battery. 

12BH611-GR, from a top view, having the snap on the right side:

- Pin 1 is used for the negative side (upper)
- Pin 2 is used for the positive side (lower)

But in order to work with Device/Battery, it should be:

- Pin 1: positive side (lower)
- Pin 2: negative side (upper)

I went ahead and changed it in my local copy, and verified that everything worked the way it should after swapping the pin numbers.

evanshultz, 2018-11-26:
Thanks for reporting this potential issue. I see this is your first post here, and I'm sorry it's because of a problem but thanks for taking the time to let us know!

The datasheet is always helpful so we can all understand the problem. I found https://www.mouser.com/datasheet/2/209/EPD-200766-1274481.pdf and it shows this:
[image]

Here is the footprint in our current library:
[image]

Both have the snap on the left side and the pinout matches with negative on the top.

Also note that pin 1 is negative.

And here the schematic symbol that would go with this footprint you mentioned above (Device:Battery_Cell is the same pinout):
[image]

Pin 1 is positive. So that's the issue. The pin numbers of the pins on this footprint need to be swapped.

I checked all other footprints with a "+" marked in them (I didn't check unmarked ones) and pin 1 is positive elsewhere, so this is the only broken footprint I found.

Sorry that you ran into this issue but thanks again for reporting it. Are you willing to make a PR to fix this?

frarees, 2018-11-26:
Hey @evanshultz, thanks for reaching out. Really appreciate your time and your response.

I will send a PR later today including the pin swapping.

PR #1114 (2018-11-27): Fix 12BH611-GR pad numbering
Fixes #1113
commit 199f6918 (2018-11-27): Fix 12BH611-GR pad numbering (#1114)

* Fix 12BH611-GR pad numbering

Fixes #1113

* Additional improvements to ensure compilance

- Pad 1 set to rectangular
- Pad 1 specified as footprint anchor

* Further improvements to ensure compilance

- Unrotate pin 1
- Silkscreen layer requirements reviewed
- Fabrication layer requirements reviewed

* Reposition labels
closed 2018-11-27 by evanshultz