KiCad/kicad-footprints #1120: Fiducial questions

antoniovazquezblanco, 2018-11-29:
Some of those fiducials in the lib are really massive. Should they still exist? What are they used for? Do they conform to any standard?

Could you please explain to me why there are fiducials in bottom layers with silk refdes in the top ones? Shouldn't bottom layers use mirrored top layer fiducials? Why do those exist?

poeschlr, 2018-12-04:
These might simply be there for legacy reasons. I would suspect that they where unchanged in the lib for a very long time. So if you have the time to update them to better standards go ahead. 

As far as i am aware it is kind of untypical in todays environment to include them at all. At least on the user side. The fabs might add them outside of the board area. (They will however use their own tools and specialized "footprints" for this.)

herostrat, 2018-12-11:
I just want to chime in on the use of fiducials:
I am currently in the process of getting a board of mine fitted in a fab.

Most fabs I spoke to and that sent me their guidelines prefer and often demand fiducials.

Also look into projects like: https://github.com/openpnp/openpnp
Optical recognition and recalibration is an important step and fiducials are used.

herostrat, 2018-12-11:
Yeah no problem, would love to help out.
I was a quite busy the last months with my masters so my contributions were not as much as I hoped.
Currently there is a bit of downtime to help out.

One important addition for me was already added from someone else (https://github.com/KiCad/kicad-symbols/pull/1131). With it you don't need to fix the footprint in order to get a correct POS file as you all correctly found out in https://github.com/KiCad/kicad-footprints/issues/1045.

I will accumulate the data and see what makes sense (and what I am allowed to publish for reference)

herostrat, 2018-12-17:
I just called two fabs.

The unanimous assertion is that for nearly all cases they prefer and want round fiducials in the dimension of which I wrote/the forum post produced.

Silkscreen marks are not recommended at all and should not be used.

There are a view cases in which the soldermask opening can and should be different (in most cases quadratic) if otherwise the orientation of the board is not discernible because of symmetry.

I do not think bottom layer fiducials are needed at all.

---

I propose to delete _all_ fiducials except:

 - Fiducial_0.5mm_Dia_1mm_Outer
 - Fiducial_0.75mm_Dia_1.5mm_Outer
 - Fiducial_1mm_Dia_2mm_Outer

I do not know how you guys handle backwards compatibility?
No symbol has an atomic footprint to any fiducials associated to it.

If it is ok for, I will create a PR removing the other fiducials.

evanshultz, 2018-12-17:
@herostrat 
This is great research! Thanks for pulling this all together.

Note that Figure 3-13 of IPC-SM-782 shows some other fiducial styles and their sizes. I've never seen it used and am OK removing them, though.

I've typically used a 1mm fiducial everywhere but have been asked by some CMs to use a 1.5mm fiducial (and it's referenced in several of the documents/sites above), so I'd also include that pad size.

Also, should we include the 3R mask diameter instead of just 2R? My own experience is that using the smaller size is probably the most common by far, and if it's requested we could add it later.

Lastly, I think the naming could even be simplified a bit. How about this (adding the one I suggested):
- `Fiducial_0.5mm_Mask1mm`
- `Fiducial_0.75mm_Mask1.5mm`
- `Fiducial_1mm_Mask2mm`
- `Fiducial_1.5mm_Mask3mm`

With regards to deleting them, I would think perhaps they get moved to an obsolete folder (it doesn't yet exist for footprints) for 5.1 (unless we decide to push that back to v6). That gives us a chance to get feedback from users that are impacted and tweak our process.

poeschlr, 2018-12-17:
moving them to obsolete might indeed be the best option. This allows users who want them to easily get them (getting them from git might be a bit too much for some of our userbase)

herostrat, 2018-12-18:
I like the idea of also including the 3R variants. I personally prefer them if there is enough space.

That would give us:
 - Fiducial_0.5mm_Mask1mm
 - Fiducial_0.5mm_Mask1.5mm
 - Fiducial_0.75mm_Mask1.5mm
 - Fiducial_0.75mm_Mask2.25mm
 - Fiducial_1mm_Mask2mm
 - Fiducial_1mm_Mask3mm 
 - Fiducial_1.5mm_Mask3mm
 - Fiducial_1.5mm_Mask4.5mm

The description would be something like this:
Circular Fiducial, %f bare copper, %f soldermask opening; (additional note)

With additional note beeing for:
 - Fiducial_0.5mm_Mask1mm: Level C
 - Fiducial_0.75mm_Mask1.5mm: Level B
 - Fiducial_1mm_Mask2mm: Level A
 - Fiducial_1mm_Mask3mm: Recommended

herostrat, 2018-12-18:
The updated fiducials are now merged.

Thanks for the fast responses!

closed 2018-12-18 by evanshultz