KiCad/kicad-footprints #1153: Some connectors in Connectors_USB not appearing in cvpcb

cryarchy, 2018-12-08:
Kicad V5.0.2

Some of the connectors in the Connectors_USB do not appear in the list of footprints in the rightmost pane of cvpcb when all but the *by library* filter is turned on and the *Connectors_USB* library is selected on the leftmost panel.

herostrat, 2018-12-12:
Hey cryarchy,

I can not recreate this problem on my installation.

Although I think this is more an issue with the kicad source and not the library (and should be reported here: https://bugs.launchpad.net/kicad/+filebug) you could post some screenshots of your problem so we can see if I can help you.

Firstly, let's see the image of your librarybrowser of the usb connectors:
[image]

Then a image of this:
[image]

Please make sure to include the buttons on the top (menubar).

cryarchy, 2019-01-17:
Here is what I have:
[image]
[image]

myfreescalewebpage, 2019-01-17:
And ?

herostrat, 2019-01-17:
Could you point me what footprints can not be seen?

I cannot see where the problem is :)

myfreescalewebpage, 2019-01-17:
Seems you want to assign a footprint, isn't it ? Are you doing this to generate your netlist ? Are you creating a new symbol ? If you can share more contextual information it's good to understand your issue.

cryarchy, 2019-01-18:
> Could you point me what footprints can not be seen?
> 
> I cannot see where the problem is :)

Your list has 24 entries while mine has only 13. About 11 footprints are missing. Is this to be expected?

herostrat, 2019-01-18:
> Your list has 24 entries while mine has only 13. About 11 footprints are missing. Is this to be expected?

Aha ok, this is the problem.
The library that comes with a version of KiCad is a tagged version of the footprint, symbol and 3d model repos.
That means they do _not_ get updated after this for this specific KiCad version.
This has to do with compatibility and stability.

If you want to use the streamingedge version of those repos you have to download them and among other things, change the path to the libraries in KiCad to them.

Rene has *very good* and in depth tutorials written for advanced library management.

You can find them in the official forum (where further questions regarding this problem are more appropriate)
https://forum.kicad.info/

This FAQ will answer a lot of your questions:
https://forum.kicad.info/c/faq

Specifically those threads are what you are looking for:
https://forum.kicad.info/t/library-management-in-kicad-version-5/14636
https://forum.kicad.info/t/how-can-i-install-a-specific-version-of-the-footprint-library/9269

For assigning footprints (although I do not know if you had a problem there)
https://forum.kicad.info/t/how-can-i-assign-a-footprint-to-a-symbol/

cryarchy, 2019-01-18:
@myfreescalewebpage Since it is obvious that this is to be expected, I guess this is no longer an issue and can be closed. @herostrat Thank you. You've been of great help. I'll check out those resources.