KiCad/kicad-footprints #1163: CRLF vs. LF: MountingHole_2mm and MountingHole_2.1mm

aewallin, 2018-12-13:
1438595d3987a6df81d4de376d6bed3544c83672 seems to cause errors for me:

$ git diff
warning: CRLF will be replaced by LF in MountingHole.pretty/MountingHole_2.1mm.kicad_mod.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in MountingHole.pretty/MountingHole_2mm.kicad_mod.
The file will have its original line endings in your working directory.

I'm not sure what is going on. I'm on Ubuntu 16.04LTS.

fauxpark, 2018-12-13:
Looks like Travis picked up on the CRLFs: https://travis-ci.org/KiCad/kicad-footprints/builds/426012316
But it only produces a warning, not an error, so you get a green tick mark on the commit. Considering the use of the word "must" as opposed to "should" both in the check scripts and [the KLC itself](http://kicad-pcb.org/libraries/klc/G1.7/) maybe it should be an error.

poeschlr, 2018-12-13:
the autocrlf thing sadly does not help if users add stuff directly via the browser instead of going through git. When using a proper git client it should work as intended.

aewallin, 2018-12-14:
Is this something I can/should fix in my local repo?
It now prevents me from "checkout my_old_branch" before committting or stashing the CRLF modifications??

aewallin, 2018-12-14:
Thanks, resolved now.
I had to delete the offending files before I could do a "pull upstream master"

closed 2018-12-14 by aewallin