KiCad/kicad-footprints #1195: How t oname variant on existing SC-59

Misca1234, 2018-12-19:
There exist a foot print SC-59

[image]

However the component AH1806
https://www.diodes.com/assets/Datasheets/AH1806.pdf

Says it support SC-59

[image]

But it is numbered in a different way.

Should I create a new foot print, fi so, how to name it so others foot print filter SC*59* wont catch it, that is, like "DIODES_SC-59" ?

or should we, not so desirable, renumber the pins in the symbol so it will match the existing foot print ?

poeschlr, 2018-12-19:
I would suggest we add some form of suffix to make it obvious that the footprint is for some alternative pad numbering scheme.

evanshultz, 2018-12-19:
https://github.com/KiCad/kicad-footprints/tree/master/Package_TO_SOT_SMD.pretty shows a couple possible options:
1. Add a `_Reverse` suffix since the pin numbering is reversed.
1. Add a `_TabPin2` suffix (it's perhaps not really a tab, but it's similar to SOT-223 in this regard).

I think KiCad pinout does not have to match the vendors. For example, you can find similar variations in SOT-23 transistors from different vendors. That being said, with a transistor symbol you don't exactly need the pin numbers but for AH1806 the pin numbers are more important.

I'm not finding this clause in KLC v3 on the website, but KLC v2 section 7.6 says `Footprints should not be duplicated to match a different pin ordering. This is to be handled in the symbol libraries.`. See https://github.com/KiCad/kicad-library/wiki/Kicad-Library-Convention. Maybe that's invalid now? It that clause is still relevant, it would mean change pin numbers on the symbol for this part so it matches the existing SC-59 footprint. That is bound to be weird when compared with the datasheet, though, so perhaps this clause was intentionally removed in KLC v3.

Misca1234, 2018-12-19:
But wont the symbols which have a FP = "SC*59*" hit these ?

evanshultz, 2018-12-19:
The symbol for AH1806 in SC-59 package would call out the new footprint, right? And because the new footprint would have a unique name it would not affect the default footprint for any other SC-59 symbol already in the lib.

But you're talking about the footprint filter. Even if it matches it's going to be obvious there's something special about a footprint that has `_Reverse` at the end. The AH1806 footprint filter could even be `SC*59*Reverse*` so ensure it only allows the proper footprint.

Misca1234, 2018-12-20:
No, I meant already existing (cant find one actually) or further symbols that use the "original" SC-59 foot print, if those set their foot print filter to "`SC*59*`" wont they hit SC_59_Reverese ?

poeschlr, 2018-12-20:
The footprint filters can never be made such that they do not filter false positives. There is a reason why we require the footprint field to be filled out in most cases. If users ignore the pre filled footprint field then it is up to them to know if the footprint fits their need.

Misca1234, 2018-12-21:
then it will be 
SC-59_Reverse

closed 2018-12-27 by Misca1234