KiCad/kicad-footprints #1211: Two sided keepout area

herostrat, 2018-12-24:
Hey guys,

I am currently designing PCB edge footprints for m.2.

I have two questions:
 - how do we handle two keepout areas for to different sizes? I found https://github.com/KiCad/kicad-library/issues/1443
 - Is there a way to make the hatched line automatically?

On a sidenote:
 - are graphical polygons without a total fill planned? This would be my first idea to look for different filling options.

evanshultz, 2018-12-25:
I have two answers (for what it's worth):
- There is no way to differentiate them clearly, so if the areas don't touch the text can relatively easily differentiate them. If they do overlay, then you have to be creative (maybe crosshatch in opposite ways?).
- No "manual" way, but see https://github.com/pointhi/kicad-footprint-generator/pull/242 for scripted footprints. This isn't merged yet and may have a ways to go? I'd done it by hand with copying and using the Move Exact feature but it was moderately painful.

I don't know about polygons. Perhaps check the dev mailing list. v6 is supposed to have a major revamp of constraints (design rules) and keepouts are one of the many, many constraints that are needed.

Misca1234, 2018-12-26:
FYI
Beside the polygon, there is one for circle as well for scripted foot prints (not merged yet)
https://github.com/pointhi/kicad-footprint-generator/pull/243

herostrat, 2019-01-15:
Thanks for the help.

I will make a wishlist issue on launchpad to add a front and back keepout area option in the layers.
As long as this is not implemented the workarounds are named, although I do not think they are very elegant

closed 2019-01-15 by herostrat