KiCad/kicad-footprints #1231: QFN-44-1EP_9x9mm_P0.65mm_EP7.5x7.5mm datasheet with insuficient data

poeschlr, 2018-12-29:
The datasheet of this footprint has some missing dimensions and definitely missing tolerance information. The only thing well dimensioned is the suggested footprint. However that footprint should then definitely be marked as a manufacturer specific one. At the least the suffix PullBack must be added.

This would require changing the footprint assignment of P8X32A-M44

evanshultz, 2019-01-02:
I find the following to be missing:
- Package body tolerance (on the 9mm dimension), but this could be accepted as a nominal dimension
- Lead width (and tolerance or just in min/max)
- Tolerance of the 0.1mm lead-to-package-edge dimension

Is there more that we would need? I'll contact Parallax once we settle on what we need.

poeschlr, 2019-01-02:
the lead width is the thing that is mainly missing. (but also the tolerance for the pull back as i hightly doubt this to be a reference messurement.)

Body sizes are quite often given with 0 tolerance.