KiCad/kicad-footprints #1243: sub d footprints use pin number 0 for shield (or mounting) connection

poeschlr, 2018-12-30:
This should be fixed to SH or MP

poeschlr, 2018-12-30:
The symbols specifically designed for sub-d connectors however also use pin number 0.
This means this can wait till the next major release.

hildogjr, 2018-12-30:
Just for clarification:
symbols are using `SH` pin, but footprint using `0` pin for shield.
[image]

poeschlr, 2018-12-30:
@hildogjr this is only true for the generic symbols. The specialized symbols for sub d connectors use 0. Meaning use these symbols and everything should be ok for now.

poeschlr, 2018-12-30:
For reference the screenshot of the current db9 connector symbol
[image]

poeschlr, 2018-12-30:
The library is called Connector (It holds every connector symbol that does not fit into Connector_Generic)

Is it possible that your local lib is from sometime before the v5 release? (a late nightly perhaps)

poeschlr, 2018-12-30:
Updating kicad does not update your library tables. Meaning if you had nighlies before the v5 release then you will have a library table older than v5. (The lib was only declared stable after rc2 of version 5)

hildogjr, 2019-01-06:
Thx
I replaced my definitions by
https://raw.githubusercontent.com/KiCad/kicad-symbols/master/sym-lib-table
and works fine.
(Yes, even with `apt purge kicad*` and deleting `~\.kicad\`, this configuration was kept).

tijuca, 2019-01-10:
Basically Offtopic ...
> (Yes, even with `apt purge kicad*` and deleting `~\.kicad\`, this configuration was kept).

This is the expected and wanted behavior of apt or respective tools. And if some package is touching anything directly in the users home folder while apt or dpkg is running this is a serious issue.