KiCad/kicad-footprints #1250: Allow PCB to have unrouted netlist actions

codemonkeyrawks, 2019-01-02:
I would like to see in KiCad the option to allow unrouted PCB parts like holes in boards to not be removed each time I have to update the netlist actions in schematic view to PCB export.

evanshultz, 2019-01-02:
This repo is to be used to discuss the libraries, not the tool itself.

You will find https://forum.kicad.info/ to be a good resource for discussing KiCad and how it works, and 
https://launchpad.net/kicad is the place to go with requests or bugs.

For this specific item, you can lock the footprints so they don't get ripped up, though there are still caveats with the locking approach. Updating the netlist is more smooth in the 5.1 nightlies, BTW.

I'm closing this now but please continue to chase down this issue and come back soon with topics involving the official libraries!

closed 2019-01-02 by evanshultz

poeschlr, 2019-01-02:
I feel nice right now so i give you a tldr of what to do.

Option 1: Include everything in the schematic. We have a mechanical symbol lib for a reason.
Option 2: Lock the footprint you have added from within pcb_new. This will prevent it from being removed (but also lock its position)