KiCad/kicad-footprints #1264: Zero orientation of oszillator lib is wrong

poeschlr, 2019-01-05:
The script creating oscillator footprints creates them in the wrong orientation.

evanshultz, 2019-01-06:
https://github.com/KiCad/kicad-footprints/issues/155

closed 2019-01-06 by evanshultz

poeschlr, 2019-01-06:
The issue you linked is about polarity markers not about the zero orientation. Are you sure this is a duplicate? (I think the other issue might have already been resolved.)

evanshultz, 2019-01-06:
The zero orientation was mentioned in the post from the old repo that resulted in #155. Since these will surely be scripted, perhaps one issue is better than multiple? I'll clearly add this to the old issue but if you want to reopen this issue or do something else please go ahead. Sorry if I was too hasty.