KiCad/kicad-footprints #1302: About handsoldering version of footprints

gmsotavio, 2019-01-14:
I'm designing a SMD crystal footprint, but I realized that the KiCad library usually have both the standard footprint and the hand-soldering version. I have been following the datasheet for the stardard footprint. But I do not know how much I need extrapolate the dimensions of the pads to be considered a hand-soldering version. Is there some rule or convention that main KiCad librarians use to a footprint be eligible as hand-soldering?

evanshultz, 2019-01-15:
Good question.

There isn't any particular method or rationale that I've seen. If you have some thoughts please go ahead and share them.

My only suggestion is that enlarging pads for hand soldering is likely to happen for small parts where a fine tip iron would be used. The tip of the irons I'm familiar with are <1mm (0.5mm or less is common for me) so enlarging by 1-1.5mm seems reasonable. I'm comfortable soldering on "regular" pads for 0402 and 0.5mm pitch, though, so we should get the opinion of other folks too.

herostrat, 2019-01-17:
You can find additional information and discussions about the handsoldering versions here:
https://github.com/KiCad/kicad-footprints/issues/497

poeschlr, 2019-01-22:
My definition for handsolder footprints is a slight increase of the pad size towards the outside (the toe fillet for most devices)
The amount of toe increase depends a bit on device type. One needs a lot less for things like smd resistors when compared to a qfn for example. (If it is even possible to have a handsolderable version of the later.)

myfreescalewebpage, 2019-01-26:
@gmsotavio do the previous responses answers your question ?

Thanks
Joel

gmsotavio, 2019-02-09:
@myfreescalewebpage Yes. Thanks folks, for all answers. @evanshultz provided a good suggestions.

closed 2019-02-09 by gmsotavio