KiCad/kicad-footprints #1307: Templates vs Footprints

herostrat, 2019-01-17:
I think in issue https://github.com/KiCad/kicad-footprints/issues/864 the problem stems from a misunderstanding of the use cases of templates and footprints.

In my understanding:
 - Footprints are used to add components and modules to a PCB. If the component is a PCB itself, it can be seen as a daughterboard mounted to the mainboard that is the project PCB

 - templates are used if something gets mounted to the module, in other words if the projects PCB acts as a daughterboard and gets mounted to the motherboard that is the templates outline.

To avoid confusion:
A arduino footprint would be mounted on the PCB, e.g. this arduino nano breakout board https://store.qkits.com/arduino-nano-breakout-board.html

A arduino template would be a shield, e.g. the https://store.arduino.cc/arduino-ethernet-shield-2

Is my understanding correct?
Is there a official definition of this?
Should we add this definition to the KLC and the docs to avoid confusion?

----

In order to track progress, here are working points I plan to update:

 - [ ] Verify the differences and usecases of "footprints" vs "templates", define rules with the maintainers
 - [ ] Update the documentation accordingly
 - [ ] Recheck if all templates and footprints are correctly designed in such a manner
 - [ ] Monitor and push the template functionality
   - Folder for templates
     https://bugs.launchpad.net/kicad/+bug/1808011
   -

herostrat, 2019-01-17:
Ok interesting.
Thats unfortunate, because that means that the confusion remains - at least I think it is not clear what is meant for what.

----

I do not understand what you mean with block?
You mean something like a blueprint?

calebreister, 2019-01-17:
As a first-time contributor working on [#1304](https://github.com/KiCad/kicad-footprints/pull/1304), I can confirm that the particular use cases can be confusing. Perhaps it would be a good idea to add a section on templates to the KLC.

poeschlr, 2019-01-17:
There simply is no clear cut rule as to what should be a template and what should be a footprint. (For things that can be a template)

The main problem might simply be that we use templates in a way they where never really intended to be used. This is because there is no way to define a block (or blueprint) in kicad which would in most cases be the better option. So we use templates as such.

herostrat, 2019-01-18:
Ok that is at least a bit more understandable.

As it stands now, templates are completely useless.

Is there any discussion as to what should templates, blueprints and module footprints do where you can point me at?
I understand that the current solution is a workaround, but imho without rules we get suboptimal results where users use all tools wrong, each with his/her own understanding and therefore wrong usage.

---

I found this issue which seems to be the wishlist item for blocks/blueprints:
https://bugs.launchpad.net/kicad/+bug/1797683

---

Until this gets (if ever :D) implemented, we should make the best of what we have and include clear rules into the KLC.

---

Proposal (that seems the most logical to me, so please leave feedback):

 - Templates:
   1. pcbs with standardized dimensions (m.2 cards, pcie cards, euroboard, IEC60297, ...)
   2. pcbs for shields of common modules (raspberrypi shield, arduino shield, ...), because they also have a specific size 

 - Footprints:
   1. Everything that gets mounted to the pcb one is designing (DCDC modules, arduinos, ...). The design has in mind the way those modules have to be mounted, e.g. the arduino uno has to be mounted upside-down, because the pinheader are already soldered on delivery

 - Blueprints/Blocks:
   This has to be defined and specified in KiCad imho, although the librarians should have a say in this too. It could well be that this could replace and extend the templates. Afaik there is no detailed specification of this as of yet. I think a hack without discussion as it seems in the launchpad thread does not add anything useful...

evanshultz, 2019-01-19:
I wouldn't say templates are completely useless. There are several includes with KiCad and they are useful IMO.

At http://docs.kicad-pcb.org/stable/en/kicad.html#_project_templates, you will find the only mention of templates in the documentation. I quote:
```
Using a project template facilitates setting up a new project with predefined settings. Templates may contain pre-defined board outlines, connector positions, schematic elements, design rules, etc. Complete schematics and/or PCBs used as seed files for the new project may even be included.
```

Notably, the documentation is stale all around and may not reflect the actual capabilities and real-world use of KiCad. But... for what it's worth there you go.

It seems to me that if the project being designed in KiCad is a "motherboard", which will use some module, then a footprint for the module is the right choice. But if the project being designed _is_ the module, so you want a canvas on which to design circuitry, then a template is what you want. There are valid reasons for having, for example, a M.2 2280 with B key in both a footprint and a template.

Here are my thoughts on the three issues/PRs noted above:
- https://github.com/KiCad/kicad-footprints/issues/864: A decision needs to be made to mirror or not as a rule for all templates.
- https://github.com/KiCad/kicad-footprints/pull/1124: Cards are templates and sockets are footprints.
- https://github.com/KiCad/kicad-footprints/issues/1064: I am still not clear why a template is not suitable, but it's certainly possible I'm a bonehead.

myfreescalewebpage, 2019-02-13:
Just read the whole topic and want to react about my PR #1124 because I clearly understand the difference between the template and the footprint but in my position, I'm designing a mini pcie card, currently I use the full card but I will be happy to optimise my layout to get it in a mini card format.

I will be able to change the footprint easily to change the card format, while it is absolutely not possible with templates, I will need to restart the design from scratch if I use them...

My position is that Edge.Cuts to define some "card footprints" **which are normalized** in the lib is very convenient.

Some M.2 footprints have been proposed (opened PR at this time) and will have the same issue.

Cheers,
Joel

herostrat, 2019-07-06:
I thought about this topic and the general idea of templates/blueprints a bit.
To find out what a better way of handling things is I created a forum post:
https://forum.kicad.info/t/rfc-kicad-blueprints/17926

Please feel free to add you comments and ideas.
I need the input of devs and librarians to develop a solid concept.

evanshultz, 2019-07-10:
I'm not sure what exactly you're looking for. The forum post has some good ideas and Bob being Bob. As usual. But also (I think) correctly pointing out the goal should be to add what KiCad users want first. Seth shares the blueprint for a nice addition that, based on historical precedent, won't arrive in a stable release for quite a while now. Since your forum post is entirely about what you would like, I gather you're looking towards the future?

Should this nice and valuable wishlist be at odds with the current KiCad release? While documentation is sparse, I understand the current meaning of template, as defined by the templates included with KiCad 5.1.2, to be a "vanilla" schematic and board combination with a board outline defined and some minimal circuitry added as a starting point from which a project can be built. An inherently incomplete building block meant to bootstrap a user getting started with whatever they want to make.

Is there another possible meaning? That's all I can make out of the existing templates.

So another question is: what does the official library define "footprint" and "template" to mean? While the meaning can shift, it would be nice if the existing templates are valuable to some KiCad users. And if so, would more templates add additional value? Without a definition it's hard to accept contributions to this repo.

It frustrates me why this conversation keeps coming up and never goes anywhere. An executive decision seems sorely overdue and Rene's responses above don't make any progressing toward closing this question. Somebody is always going to be unhappy but that need not paralyze any decision-making even in the face of a potential vocal minority, if there is one. A higher power can decide what a footprint and template are right now, and then when KiCad grows even more cool features the official library can evolve to suit. (Naturally, there are lots of clever and dedicated folks using KiCad and they are certainly free to do whatever they would like with their own library.) I would like the official library to define these terms and then we can accept more contributions and make KiCad users happy.

Sorry for the snarkiness.

@stambaughw 
Perhaps you can share what templates are meant for as of KiCad 5.1.2? Then the templates in the official library can represent that use case (if they don't already) and new templates fitting that use case, if contributed, can be added. Thanks!

herostrat, 2019-07-10:
The forum post was created to find out what others want/need from templates/blueprints and look if it is possible to create it. I was not aware of the spec and the goal for it to land in v6 (I think because of the choice of vocabulary for the elements).
I also wanted to see if it is possible to create or finance the development of such a feature if enough people are interested.

I share your frustration, I basically ask again and restart the topic because examples like https://github.com/KiCad/kicad-footprints/pull/1668 seem to be outside the current use of the footprint modules.
I don't want to push a decision or decide anything myself, but I think it's important to mark and define what is used for what and where to make a PR for it.
There are a lot of module PR and I don't think they become less in the future.

evanshultz, 2019-07-10:
Completely agree. Knowing the intended use for templates seems, to me, like the first place to start. If the system can then be "abused" and allow other possibilities, so much the better as long as it's helpful to KiCad users.

As the volunteers reviewing and merging contributions, the librarians need to be on the same page with what should be in the official library. So I'll keep pushing because I also see templates, as I understand them to be, growing.

And seeing the concepts you and other folks proposed and what will hopefully be coming in v6 makes me even more excited for future KiCad!