KiCad/kicad-footprints #1308: add Mini-Circuits footprints without GND-zone/vias

aewallin, 2019-01-17:
As discussed in #1133 it was agreed to add "plain" versions of the Mini-Circuits footprints that do not contain GND-zones or GND-vias.

Here is an attempt to list the footprints that are merged, but do not have corresponding "plain" (non GND-zone, non GND-via) footprint in the library.

- [ ] BK377 PL-005 #1126
- [ ] GP731 PL-176 #1136 
- [ ] GP1212 PL-176 #1258 (same as GP731 but different height package, an difference only seen in 3D)
- [ ] HF1139 PL-230 #1134 
- [ ] MMM168 PL-225 #1150
- [ ] QQQ130 PL-236 #1135
- [ ] TT1224 PL-258 #1128
- [ ] YY161 PL-049 #1137

additionally the following PRs have the same issue

- [ ] PL-012 in #1151 
- [ ] PL-079 in #1133 
- [ ] PL-247 in #1132

herostrat, 2019-01-18:
Just to understand, do you plan to do the work yourself or is this a "need help" type of issue?

aewallin, 2019-01-18:
These are not at the top of my to-do list - I may tick them off slowly one by one but don't hold your breath ;)
Please do feel free to work on this!

aewallin, 2019-01-18:
I think these suggested new "plain" footprints create a bit of an issue with the footprint-filters in the merged symbols.
The plan above is to have two symbols
- mini-circuits_<Case>_LandPattern<PL-Code> with gnd-zones/vias
- mini-circuits_<Case>  the "plain" footprint, no gnd-zones/vias

But in work that is now merged the fp-filter is set to "mini-circuits*<PL-Code>*", so it will *not* match the plain footprint!?

poeschlr, 2019-01-18:
the footprint filter for the symbol might best mention the case code.
The default footprint should of course use the full footprint name including the land pattern.

aewallin, 2019-01-24:
these now merged. closing this.

closed 2019-01-24 by aewallin