KiCad/kicad-footprints #1313: Git history went backwards

fauxpark, 2019-01-19:
I have these commits in my fork which are not in upstream:

[image]

poeschlr, 2019-01-19:
Hm i fear @Shackmeister might have some explaining to do.

[image]

Shackmeister, 2019-01-19:
@poeschlr I'm afraid he has aswell. Was catching up on my local repo and I must have accidently pushed to master. Very sorry for the trouble

poeschlr, 2019-01-19:
@fauxpark i pushed the state of your fork to the main repo.

@Shackmeister did your now remove push possibility from your local copy?
check by running git remote -v
make sure there is no legal link for the (push) option of the main repo.

poeschlr, 2019-01-19:
I think i got everything back in that was lost. (There where only 3 pull requests merged after @fauxpark pulled from the repo.)

Thanks @fauxpark for reporting this so fast. Made it possible to be fixed quite easily.

closed 2019-01-19 by poeschlr

Shackmeister, 2019-01-19:
I have now disabled push to all the kicad repos

@fauxpark @poeschlr Sorry for the mess, and thanks for cleaning it up.
Is there a reason we haven't disabled it "globally"? If it's needed sometimes for cleanup I suppose it could just be enabled when needed and otherwise left off? not trying to cover my own stupidity, but just curious

poeschlr, 2019-01-19:
What should we disable globally? You having push rights is what allows you to merge pull requests. If i turn that off for you then you can no longer merge contributions defeating the purpose of being a maintainer.

Shackmeister, 2019-01-19:
makes sense, thought the merge and push was considered 2 seperate actions (but now that I think about its makes sense its both)

poeschlr, 2019-01-19:
There is no option i can see for disallowing force pushes. I know that gerit has such a feature but i can not see one on github.