KiCad/kicad-footprints #1316: Appendix: MountingHoles vs WithMountingHoles

herostrat, 2019-01-21:
[image]

Currently researching the appendix MountingHoles for footprints for the NodeMCU modules.
I saw that there are (at least) two different ways currently used.

Reading the HandSolering/HandSolder thread makes me think MountingHoles would be the way to go, although I do not know why we have footprints without the MountingHoles at all?

herostrat, 2019-02-13:
See: https://github.com/KiCad/kicad-footprints/pull/1066#issuecomment-462539470

evanshultz, 2019-02-13:
By "appendix" you mean the suffix of the symbol, I assume.

Probably a lot of the discrepancy is because these modules have been contributed by many different people over a long period of time with changing library requirements. A nice clean up would be good, and if we decide to split off the headers that would be a reason to go through all of them.

I would agree to keep the `_MountingHoles` suffix if the footprint includes them, or else don't include this suffix.