KiCad/kicad-footprints #1320: Mean Well IRM-03 wrong footprints, maybe more due to docs

EngiBob, 2019-01-21:
Hi,

The main problem is IRMs docs show things from different bottom view angles/rotations.

If you pull up the Kicad footprint  Converter_ACDC_MeanWell_IRM-03-xx_THT it exactly mirrors the docs at https://www.meanwell.com/Upload/PDF/IRM-03/IRM-03-SPEC.PDF

If you look at an actual picture of the IRM-03 series you can see the mechanical drawings are actually upside down on the document (not all models are upside down, some are rotated around the y axis not z axis):

[image]

I just did a project using 3 modules, IRM-03, IRM-10, and IRM-15 class.  There is no 15 in Kicad so I did my own with the help of product pictures (it is upside down too), IRM-10 is OK it looks like, and I'm doing IRM-03 right now...  I've done two other projects with modules I can't remember at the moment and one of them I even  swapped my output +/- and fried a test board due to the docs all being different.  Where I can send the ones I've done for possible inclusion?

EngiBob, 2019-01-21:
Also an error in the spacing, has 7.18 instead of 7.78.

EngiBob, 2019-01-21:
I redid the library version of IRM-03, except for the position of the triangle for neutral.  I also added the holes (for ventilation?).   

https://gist.github.com/EngiBob/ddb95417a3fb4fbaaf60a70efe0b4cd3

EngiBob, 2019-02-19:
They soldered up my board without any issues, got them back today and the footprint works great.

antoniovazquezblanco, 2019-02-19:
@EngiBob Could you please send a pull request to fix the issues in the official lib? It would be of great value to us.

evanshultz, 2019-02-19:
@EngiBob 
Sorry to hear about your troubles and that there were due to errors in our library.

Since there's a lot going on above let me work through all the MeanWell IRM series we have now.
- IRM-02: Both THT and SMD symbols and footprints look OK to me. Note the THT symbol uses the pinout AC/N, AC/L, -V, +V, and MeanWell doesn't give pin numbers for THT packages.
- IRM-03: SMD looks OK in both symbol and footprint. The THT footprint is flipped but we have no THT symbol.
- IRM-05: No symbol in the library (and only THT package) so unable to verify pinout. Pin placement matches datasheet.
- IRM-10: No symbol in the library (and only THT package) so unable to verify pinout. Pin placement matches datasheet.
- IRM-20: The pinout here is AC/L, AC/N, -V, +V; annoying different than above. But the symbol and footprint match and pin placement matches datasheet.

Since this report is specifically about IRM-03, and the official library has no symbol, unfortunately it seems the symbol creation didn't involve validating the footprint.

Proposed changes:
1. Flip IRM-03 THT footprint.
1. Add IRM-03, IRM-05, and IRM-10 THT symbols to avoid the need for external symbols which could cause problems for KiCad users. Match the existing footprint pinouts to avoid anything already in the library getting changed.

Other notes:
- Many wrong symbol pin names ("-Vout" instead of "-V", for example). Could be fixed now.
- SMD footprints are missing a silk pin 1 mark. Could be fixed now.
- Footprint filters don't capture THT vs SMD footprints. Could be fixed now.
- It's annoying that the pinout isn't consistent for the THT parts, but that's not something we will do now since it would break schematics.

evanshultz, 2019-02-26:
@EngiBob 
Thanks for this report. I've submitted 2 PRs, shown above, that will fix this issue. If you'd like to verify my fixes are correct that would be welcome. Thank you.

closed 2019-03-21 by poeschlr