KiCad/kicad-footprints #1322: Switch is the function not Button.

robkam, 2019-01-22:
Button_Switch_*.pretty should be renamed Switch_Button_*.pretty, see KLC F1.1 - Footprint libraries are categorized by function.

fauxpark, 2019-01-22:
I think the name is meant to be interpreted like "Button/Switch", but slashes are illegal characters in folder names, so an underscore is used instead.

fauxpark, 2019-01-22:
Let me put it another way. The libraries are for "Buttons & Switches" (or "Switches & Buttons" if you prefer). Neither take precedence over the other.

robkam, 2019-01-22:
That's wrong, as it's mainly switches. What about "button cells"?

fauxpark, 2019-01-22:
If you mean these:
[image]

Then they would go in `Battery.pretty` (well, mainly the holders/receptacles for them, batteries themselves are not generally part of a PCB design AFAIK)

evanshultz, 2019-01-22:
@robkam 
So you're interpreting the word "button" to mean the thing the user touches to activate a switch? So the red and black plastic things in the picture below?
[image]

And the word "switch" is the electronic part that has a schematic symbol representation?

I think I understand what you're saying, but I want to be sure I do first.

robkam, 2019-01-22:
Yes that's right. Switch is the primary electronic function. The sub-function might be button, DIP, slide, toggle, etc. 

(Rotary switches are at Rotary_Encoder.pretty).

evanshultz, 2019-01-22:
It's reasonable then to call the library just `Switch.pretty` and that makes sense to me. The switch is the electronic part.

However, that would mean breaking all schematics which use a symbol in this library, with the belief that the recovery system isn't robust, but even if the symbols are recovered from a re-named library it will still be a hassle for users that build schematics with this library. And it's just not worth it.

robkam, 2019-01-22:
It's not named as per the KLC and will have to be fixed sooner or later. The longer it's left the more broken designs there will be to fix when it does get changed.

poeschlr, 2019-01-22:
To me a switch is a device that keeps its state whereas a button is something that returns to a zero state. (But this might be because in german we have taster and schalter and these are definitely not interchangable.)

Even if we decide this to be change we can only do it with version 6. (Only major releases are allowed to bring design breaking changes.)

robkam, 2019-01-23:
That makes sense. After looking at the  German Farnell element14 page for "taster" vs. the English one for "pushbutton switches". In English `Button_Switch` would be better described as e.g. `Switch_Momentary`. 

But e.g. DIP slide switches are never buttons, except that's the library they are in.

Until version 6 any new switch footprints can go in `Switch.pretty`.

poeschlr, 2019-01-23:
Splitting the lib now (for only new footprints) will only result in adding further confusion. 

---

From the oxford dictonary:
- Button: A small device on a piece of electrical or electronic equipment which is pressed to operate it.
- Switch: A device for making and breaking the connection in an electric circuit.

Wikipedia definition for push-button:
A push-button (also spelled pushbutton) **or simply button** is a simple switch mechanism for controlling some aspect of a machine or a process.

From these i would gather that we are not completely wrong. Meaning i see no particular reason why we should rush this now. We can discuss this once v6 development is opened.