KiCad/kicad-footprints #1343: Several script-generated DFN footprints - Silkscreen pin 1 mark incorecct

JacobEFO, 2019-01-26:
I discovered a minor issue within the kicad-footprint-generator that resulted in incorrect pin1 markings, on several DFN footprints currently found in the library.
Refer to the issue at: https://github.com/pointhi/kicad-footprint-generator/issues/278

The new footprints found at kicad-footprint-generator have slightly different dimensions on solder pads. I have a branch (https://github.com/JacobEFO/kicad-footprints/tree/DFNfootprints) ready for merge that updates the silkscreen (and reduces pad size slightly as per the script and datasheet dimensions).
If interested I can open the request.

evanshultz, 2019-01-28:
The old pads were built per the datasheet. Scripted footprints are newer and use IPC calculations to derive pads based on the package drawing; this may or may not match the datasheet footprint.

For scripted footprints, they SHOULD NOT be changed. IPC-compliant footprints are the way to go in a general-purpose library like ours. Mismatches between the datasheet and our footprints are ignored.

If you are bringing up a different issue please elaborate.

JacobEFO, 2019-01-28:
This issue is closely related to the silkscreen issue discussed on https://github.com/pointhi/kicad-footprint-generator/issues/278
Which was totally a mistake on my behalf.

However the current footprints are not IPC compliant (which the script generated DFN packages are). Maybe it's worth updating the footprints with the script generated ones (IPC compliant), the ones I linked to.

poeschlr, 2019-01-28:
The main work for getting further footprints generated is finding the datasheets. A tip on that: search the lib which symbols point to the footprints. This is mostly because a lot of footprints do not have a datasheet assigned. Searching for datasheets by footprint is not really a possibility so the best option is to search by component (symbol)

Another tip is to find the original pull request that added a component. (This is not as trivial as you need to follow the trail of renamings through the git history.)

If you are overly motivated then you can even check if the symbols who are assigned a particular footprint really fit that footprint. (I discovered that especially EP sizes are not always correct for symbols meaning you might need to add additional footprints.)