KiCad/kicad-footprints #1420: Are pin 1 markings always necessary?

evanshultz, 2019-02-19:
@KiCad/librarians 
KLC is clear about pin 1 marking at http://kicad-pcb.org/libraries/klc/F4.2/ and http://kicad-pcb.org/libraries/klc/F7.2/ and http://kicad-pcb.org/libraries/klc/F7.3/.

One example is given in F7.3 where a unique pin 1 pad shape is not needed if the part is non-polarized (the example is a THT resistor footprint). But the other two sections mentioned above would seem to require pin 1 markings on all footprints.

I think there are already known cases where this doesn't make sense. Two examples are audio jacks, as the schematic symbols have already been changed to pin names instead of pin numbers (so footprints need to use pin names as well), and external cards (CF, SIM, etc.) where pin names allow easier mapping between many vendors in case their pin numbering isn't aligned but there are standardized/required signals.

Here are two open PRs which are intertwined with this issue:
- https://github.com/KiCad/kicad-symbols/pull/592
- https://github.com/KiCad/kicad-footprints/pull/598
- https://github.com/KiCad/kicad-footprints/pull/1145
- https://github.com/KiCad/kicad-footprints/pull/1416

Am I right that KLC does not currently seem to allow a lack of pin 1 when we've already decided to allow it colloquially? If so, how should this be captured in KLC?

I have advocated for pin T to be marked on audio jack footprints. However, I do not think a pin mark is always necessary because some footprints have a unique shape which, in combination with the silk and fab layers, indicates proper insertion. Having one unique pad shape or marking any pin in silk/fab does not add clarity. Please see the cases linked above for some screenshots of footprints where I believe it is not necessary to mark any pin to indicate insertion orientation.

If y'all agree I can add some proposed text below to integrate into KLC. Feel free to do the same if you like.

robkam, 2019-02-19:
#1388 is another PR where this is an issue (pots with a pin 1)

robkam, 2019-02-19:
KLC F7.3 has "**Exception:**
Non polarized parts (such as THT resistors) should not set the shape of Pin 1 to Rectangular. For these (non polarized) footprints, all pad shapes should be consistent."

evanshultz, 2019-02-19:
> KLC F7.3 has "**Exception:**
> Non polarized parts (such as THT resistors) should not set the shape of Pin 1 to Rectangular. For these (non polarized) footprints, all pad shapes should be consistent."

I explicitly noted this above.

poeschlr, 2019-02-20:
The pin 1 mark is called polarity mark in the standards. So yes resistors do not need one (Which is how we have the lib right now)

I would however still use a polarity marking for parts like audio connectors. I don't see it as a pin 1 mark for correctly inserting he part but as a help while debugging for such devices. (The same is true for displays as an example.)
This means we should simply document for non numbered pads which one gets the polarity mark. (this includes audio connectors, But also relays which IEC pin naming, ...) 
A pin that is always present in a class of devices would be an ideal candidate for being the "pin 1" (My guess would be that it is the "GND" pin for audio connectors. But it could also be some other pin.)

evanshultz, 2019-02-20:
I understand the point of a polarity (or pin 1) mark for many parts. DIPs, SOICs, e caps, etc. All makes sense. QFNs, not only have two possible orientations but four. So the part needs to be installed properly and if there's a problem someone needs to verify assembly and identify two or more pins for troubleshooting. No question there.

But for audio jacks I don't think any kind of fab, silk, or pad shape is needed. And XLRs, for example, have a pin 1 as part of the connector! Google for images which contain the color green with search strings like "pcb xlr" and "pcb audio jack" and "pcb phono" and "pcb neutrik"; if your results are similar to mine you won't see a single PCB with any kind of unique pad shape or change to the silk outline of the part. Putting pin numbers/names in silk is sometimes done for audio connectors but that's not captured in KLC.

I tried this same searching for gain potentiometers as mentioned by Rob above. I do sometimes see unique pad shapes in the images I find but I never saw anything in silk.

A common mode choke is another example where the part may fit into the PCB in two orientations but it doesn't matter as the windings are equivalent. Seeing the outline of the part in silk is adequate to know how to place it on the board and having any unique pad shape does not help. And forcing one particular orientation in production slows things down versus allowing either permissible orientation to be used.

The above types of parts, and there are surely many others, have a unique geometry so insertion is obvious. And with regards to debugging, can you give an example? Audio connectors and pots may have one naive pinout/usage which is common, but the implementation can and does vary. The ECAD system needs to have some way to match up the symbol and footprint but I believe for many parts it's not helpful to try and communicate this in fab, silk, or pad shape as a strict reading of KLC would require. Maybe you have some good counter examples?

That all being said, the overarching point I'm try to make is that requiring all of this stuff in KLC doesn't make sense for every footprint type. At least not to me. Agree? Am I wrong?

diegoherranz, 2019-02-20:
> I would however still use a polarity marking for parts like audio connectors. [ ... ] This means we should simply document for non numbered pads which one gets the polarity mark.

I'm not so sure about that one. I think that us defining which pin gets the polarity mark can be harmful.
An example:

- Someone gets a board to assemble. They didn't design it, they don't know which software/library was used, hence, there is zero chance that they check the KLC.
- They see an audio connector with a pin with a special marking. They assume that means pin 1.
- What we defined as pin 1 is different from whatever the manufacturer defined as pin 1.
- It gets assembled wrong (this is unlikely for audio connectors because it's difficult to place them on a wrong orientation but I guess it might happen on a rare case or at least raise questions from the assembler).
- Even if it gets assembled right, someone debugging the board will think that that is pin 1 when it isn't, because the manufacturer defined it somewhere else.

Because of all that, I think that if we have a marking, it has to be pin 1 or it can do more harm than good.

Having said that, even when the orientation is clear because of the shape of the part, e.g. potentiometer, I think that having the pin 1 marking is useful for debugging as @poeschlr says (it allows you very quickly to see which pin is pin 1 when probing, scoping, etc.) and it doesn't hurt having it (I would be in favour of keeping it).

Maybe we can change the exceptions on F7.3 http://kicad-pcb.org/libraries/klc/ to something like:

- Non-polarized parts, that is, parts in which the population orientation doesn't matter (such as THT resistors) should not set the shape of Pin 1 to Rectangular. For these (non-polarized) footprints, all pad shapes should be consistent.
- It the part doesn't use numbers for the pin names, and therefore there isn't a pin 1, no pin should be set to rectangular shape.

evanshultz, 2019-02-20:
Works for me, as long as F4.2 and F7.2 have a similar exception.

poeschlr, 2019-02-21:
I don't really agree with the debugging argument. First of all the "pin 1" mark does not need to be of the same shape or form as for numbered components.
Secondly debugging means you have the schematic as a reference. Not the datasheet. (Ok the datasheet will be used as a source for information to check the schematic and to check if the pin order of footpirnts is at fault for the problem encountered.)

This means to me that yes a polarity mark can still be used with these parts. (And makes sense. Especially the differentiating pad shape helps if you need to fit your measurement probe on the backside. Makes it a lot easier to identify which pin is what.)