KiCad/kicad-footprints #1434: Rename Packages_SO:SOIJ-8_5.3x5.3mm_P1.27mm to Packages_SO:SOIC-8W_5.3x5.3mm_P1.27mm

Icenowy, 2019-02-25:
The SOIJ-8 name is not a common name for this package. By searching on google with this keyboard, only materials from Microchip can be found.

However, this package is popular for 25 SPI NOR Flash, and many manufacturers do not use the SOIJ name, they just declare it as a wide version of SOIC/SOP or mention the 208mil size.

See the following:
https://www.winbond.com/resource-files/w25q128fv%20rev.l%2008242015.pdf
http://www.gigadevice.com/datasheet/gd25q127c/
https://www.micron.com/~/media/documents/products/data-sheet/nor-flash/serial-nor/m25p/m25p80.pdf
http://www.xinyahong.com/upLoad/product/month_1411/201411201249025495.pdf (This is EN25Q64)

So I suggest to rename it to follow the scheme of other wide SOICs, rather than using a vendor-specific name.

poeschlr, 2019-02-25:
SOIJ is typically a package with J lead instead of gull wing. (leads bend inwards instead of outwards) Are you sure about the renaming? (Disclaimer: I did not check the current footprint as i do not have kicad access right now. A screenshot of the current footprint would have made this a lot easier.)

evanshultz, 2019-02-25:
Here is the footprint mentioned above:
[image]

The doc referenced by this footprints is an older version of 
http://ww1.microchip.com/downloads/en/PackagingSpec/00000049BQ.pdf. However, I don't find this package type called "SOIJ" in that doc.

This footprint is used by the following symbols (just searching in Eeschema):
[image]

I agree that it would be best to change the footprint name and it's not a substantial amount of work so it could be done right away if the other librarians agree.

Icenowy, 2019-02-26:
Datasheet with the SOIJ name remaining can be easily acquired via Google.

e.g. http://ww1.microchip.com/downloads/en/PackagingSpec/00049AU.pdf

This document says "8-Lead Plastic Small Outline (SM) – Medium, 5.28 mm Body [SOIJ]", and it has no J lead mentioned.

BTW I have already used the SOIJ-8 footprint on my sample board, and it successfully get SMTed with W25Q128JVSSIQ.

poeschlr, 2019-02-26:
In that case go ahead.

evanshultz, 2019-02-26:
@Icenowy 
You can see that your link is the older version of my link. And in the newer version the term "SOIJ" has been removed. This is confirming your suggestion and we are agreeing.

Would you like to submit a PR to rename the footprint? We will also need a PR to fix all symbols currently using this footprint.

Icenowy, 2019-02-27:
Could I just rename the file? Should I keep a compatible copy of the package?

evanshultz, 2019-02-27:
When the footprint is renamed, symbols using the footprint need to then use the updated footprint name. We don't need to have a SOIJ copy of the footprint because, as you said, it's a poor description.

If the symbols are a problem, how about we fix it together? If you submit a PR to rename the footprint, I'll submit a PR to update the symbols? Does that sound OK? Or are you prepared to submit both PRs?

antoniovazquezblanco, 2019-07-18:
I am interested in this discussion. 

Up to date Microchip has a specification for a SOIJ package in their website:
https://www.microchip.com/quality/packaging-specifications
https://www.microchip.com/quality/packaging-specifications?packageFamily=SOIJ
http://ww1.microchip.com/downloads/en/DeviceDoc/8L_SOIJ_Medium_5_28_SM_C04-056C.pdf

Should this footprint be really renamed?

evanshultz, 2020-09-18:
@chschlue 
This will unlock a footprint PR and a symbol PR. IMO, perhaps Microchip uses SOIJ but I think we can use our standard SOIC footprint if it matches the package drawing. So are you OK removing the SOIJ footprint for v6 and then updating the affected symbols to use an existing SOIC footprint?

chschlue, 2020-09-18:
Yes

closed 2020-09-29 by evanshultz