KiCad/kicad-footprints #1445: Varistor hole diameters too small

martinjaeger, 2019-02-27:
I built a PCB using the footprint `Varistor:RV_Disc_D12mm_W4.3mm_P7.5mm` and realized that the hole diameter was too small. It seems like all varistors in the library have a hole diameter matching exactly the wire diameter, which is too small.

Some examples and datasheets:

Epcos StandarD series ([datasheet](https://en.tdk.eu/inf/70/db/var/SIOV_Leaded_StandarD.pdf), page 14):
- D7mm_P5mm: Wire diameter d = 0.6 mm
- D9mm_P5mm: Wire diameter d = 0.6 mm
- D12mm_P7.5mm: Wire diameter d = 0.8 mm
- D15.5mm_P7.5mm: Wire diameter d = 0.8 mm
- D21.5mm_P10mm: Wire diameter d = 1.0 mm

Bourns MOV-07DxxxK series ([datasheet](https://www.bourns.com/docs/product-datasheets/mov07d.pdf), page 2):
- D9mm_P5mm: Wire diameter d = 0.6 mm

Bourns MOV-10DxxxK series ([datasheet](https://www.bourns.com/docs/Product-Datasheets/MOV10D.pdf), page 2):
- D12mm_P7.5mm: Wire diameter d = 0.8 mm

Bourns MOV-14DxxxK series ([datasheet](https://www.bourns.com/docs/product-datasheets/mov14d.pdf), page 2):
- D15.5mm_P7.5mm: Wire diameter d = 0.8 mm

In addition to that, naming of the parts from manufacturers seems different from KiCad. Manufacturers seem to have nominal diameters of 5, 7, 10, 14 and 20, whereas KiCad uses the maximum diameter in the footprint name.

I suggest to increase the hole diameter to wire diameter + 0.1 mm for all parts.

If you agree, I can implement the changes and send a PR.

What do you think about renaming the Varistor footprints based on their **nominal** diameter instead of actual max diameter?

martinjaeger, 2019-02-27:
Just checked the hole diameter of capacitors. It's wire diameter + 0.2 mm, so that might make sense also for Varistors.

evanshultz, 2019-02-27:
Thank you for reporting what you've found using the KiCad library. Sorry you had problems.

To address your suggestion, drill holes should be 0.2mm over the lead size according to KLC (from IPC-2221).

I don't see any datasheets in the varistor footprints, so who knows why they have this drill size. The varistors do mention "SIOV", so perhaps the TDK datasheet was inspiration. Footprints were originally added at https://github.com/KiCad/Varistors.pretty/pull/1, and moved to the new lib at https://github.com/KiCad/kicad-footprints/pull/48. They mostly seem to exist the same in the current library. I see 33 with 0.6mm drill, 43 with 0.8mm drill, and 22 with 1mm drill. I checked a few and it appears 7/9mm use 0.6mm drills, 12/15.5 use 0.8mm drills, and 21.5mm uses 1mm drills.

If we were starting over, I'd support the following:
1. Picking a cross-section of part sizes and using nominal dimensions.
1. Drill 0.2mm over lead diameter, with circular pads (at least) 0.55mm bigger than the drill.

However, we have to worry about breaking all existing designs which use these footprints if they are renamed. And we can't do that. Too painful to KiCad users for little to no benefit. However, that only affects the body size and not the drill size.

Looking at Digikey, for THT disc varistors, there are several diameters which have <=100 parts and then a few with >1000 parts. So that's a pretty clear division about which body sizes are common. And I find that 5, 7, 10, 14, and 20mm are common. And the corresponding common lead diameters are 0.6, 0.6, 0.8, 0.8, and 1mm, respectively. Lead pitch is 5, 5, 7.5, 7.5, and 10mm (with 7.5mm popping up sometimes), respectively. Body thickness varies but seems to allow footprints of 5, 6, 7, 8, 9, 10, and 12mm (using 6-12 in even numbers is probably fine). Lead stagger varies much more, or isn't even given, but I have seen ~2.5mm less than the body thickness and a fixed 3.5mm regardless of body size is common. I think this matches nicely with what you found, in that our drill sizes are too small and the body sizes aren't the common sizes.

Other doc links:
- http://www.vishay.com/docs/29081/vdrs.pdf
- https://media.digikey.com/pdf/Data%20Sheets/United%20Chemi-Con%20PDFs/G_Series.PDF
- https://www.seielect.com/catalog/sei-rv.pdf
- https://us.schurter.com/bundles/snceschurter/epim/_ProdPool_/newDS/en/typ_AVTT.pdf
- https://industrial.panasonic.com/cdbs/www-data/pdf/AWA0000/AWA0000C1.pdf
- https://www.littelfuse.com/~/media/electronics/datasheets/varistors/littelfuse_varistor_za_datasheet.pdf.pdf
- https://www.littelfuse.com/~/media/electronics/datasheets/varistors/littelfuse_varistor_la_datasheet.pdf.pdf
- https://media.digikey.com/pdf/Data%20Sheets/AVX%20PDFs/VE,%20VF%20Type%20(P%20Series).pdf

So... here is what I think we can do right now:
1. Enlarge all drills by 0.2mm.
1. Add more footprints using the nominal body diameter and some number of thicknesses. This is essentially changing just the "D" part of the existing footprints (with the same, more, or less thicknesses) since I believe the lead pitch is already correct.

Does that sound good to you? Do you other proposals how to proceed?

When KiCad v6 release is approaching, if the librarian team agrees, we could remove the old (current as of today) footprints. We will have to see how things go at that time.

Lastly, varistor footprints could be easily scripted. And they should be because it will be so much quicker to maintain them, we can see where the raw part data comes from, and this is a body shape that may grow in the future so the time investment to make the script is worthwhile. I don't see a script at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts so I believe they are all hand-made now.

martinjaeger, 2019-02-27:
Thanks for the quick feedback. I agree that we should not break compatibility with current library, so we can add the footprints with new naming and still correct the hole diameters of the footprints with old names.

I saw that @antoniovazquezblanco recently [added auto generated 3D models for Varistors](https://github.com/KiCad/kicad-packages3D/tree/master/Varistor.3dshapes), so maybe he knows if there is a script for the footprints aswell?

If there is really no script existing yet, I can try to build one.

Suggestion for naming (example): RV_Disc_D10mm_P7.5mm_T4.3mm

- D is the **nominal** diameter
- P is the pitch and is the second parameter instead of third to improve sorting of similar types.
- T is the thickness, as in the datasheets. The previously used W denotes the width, which is the "maximum" D. So using W here would be confusing.

If we change the names in the footprints, we should make a copy of the 3D models with the new names, correct?

evanshultz, 2019-02-27:
Actually, Stefen added the 3D model script at https://github.com/easyw/kicad-3d-models-in-freecad/pull/226. The PR you mentioned was a small addition to the library.

I think @poeschlr should contribute to the naming. He came up with most of the footprint naming before so it should fit in his vision.

Why do names in the footprints need to be changed? I'm not sure what your last sentence is talking about. We can change the drill holes in existing footprints but no existing footprint names need to be changed.

martinjaeger, 2019-02-28:
I was referring to point 2 of your message above:

> So... here is what I think we can do right now:
> 
> 1. Enlarge all drills by 0.2mm.
> 2. Add more footprints using the nominal body diameter and some number of thicknesses. This is essentially changing just the "D" part of the existing footprints (with the same, more, or less thicknesses) since I believe the lead pitch is already correct.

Did not mean to change existing footprints, but add new (copied) footprints with new names. But the new names in the footprints should match with names of the 3D models, or not?

evanshultz, 2019-02-28:
OK. Got it. The 3D model name and footprint name should match. We can make new 3D models easily by script so that's no problem.