KiCad/kicad-footprints #1449: *BUG* Wrong Footprint Diode SMD

filewalker1, 2019-03-01:
Hello!

i found a problem with the Diode Footprint:
simple Diode library symbol has the anode labeld with pin 1, but footprint diode_smd:D_sma has anode labeld pin 2.
I am not quite sure if this is a problem with the footprint library or the schematic symbol library...

evanshultz, 2019-03-01:
Except for in the `pspice` and `Simulation_SPICE` libraries, which are only to be used for circuit simulation, all other diode symbols and all diode footprints should have the cathode as pin 1. The simulation diode is backwards because of the standard SPICE diode model's pin order.

I believe that's what has bitten you. Let's take a look.

Here is the symbol that I think you refer to:
[image]

You can see there are warnings about this diode in the schematic symbol chooser.

And here is the footprint:
[image]

You should a `Device:D*` symbol or any symbol in the `Diode` library for a diode that is for PCB design (and not simulation).

I'll leave this open for a couple days in case I've misunderstood.

filewalker1, 2019-03-06:
@evanshultz 
Thank you for your reply! you are absolutely right. I did not know there was a pspice simulation lib in there. i pressed "a" to add a Symbol an Diode was in the last-used list and i did saw the pspice hint. Sry for my false Bug Report!

closed 2019-03-06 by antoniovazquezblanco