KiCad/kicad-footprints #1450: QFN16 3x3 for NCN4555

jneiva08, 2019-03-01:
Hi,

In the [datasheet](https://www.onsemi.com/pub/Collateral/NCN4555-D.PDF) of NCN4555 says QFN16 3x3 P0.5.

The QFN-16 3x3 P0.5 EP1.75x175 don't exist. the more approximate in the repository is WQFN-16-1EP_3x3mm_P0.5mm_EP1.75x1.75mm. 

The QFN-16 should be created or can use WQFN-16?

evanshultz, 2019-03-01:
We aren't requiring this, not that I've seen anyway, but we would prefer if you create a footprint with the correct type.

This is because the WQFN and QFN likely have different heights and so the matching 3D model can reflect the actual package. Also, it avoid any confusion because the footprint in KiCad matches the package name from the datasheet.

Because the scripts easily allow making QFN footprints and 3D models it's easy to make a new footprint and so other contributors have already been doing this - making a QFN footprint even if an otherwise-identical WQFN footprint exists.

jneiva08, 2019-03-04:
Symbol: https://github.com/KiCad/kicad-symbols/pull/1601
Script: https://github.com/pointhi/kicad-footprint-generator/pull/306
Footprint: https://github.com/KiCad/kicad-footprints/pull/1458

myfreescalewebpage, 2019-03-04:
Thanks @jneiva08 for the contribution, closing the issue now.

myfreescalewebpage, 2019-03-04:
@evanshultz just realized you have assigned yourself, sorry for closing this at your place :) Anyway your answer is very nice because I have several PR in which people ask why we don't use existing approximative footprints, so now I will be able to link with this issue.

closed 2019-03-04 by myfreescalewebpage