KiCad/kicad-footprints #1452: Footprint Dual P-Mosfet DMP2160UFDB - How to commit

pboe, 2019-03-01:
Hi,
i would like to contribute a footprint for Diodes Package U-DFN2020-6(Type B)
used by e.g. https://www.diodes.com/assets/Datasheets/ds31643.pdf

Do you prefer the generic way naming it like:
DFN-6-2EP_2x2mm_P0.65mm_EP1x0.6mm
and leave the exposed pads unnamed

or more specific with name like
U-DFN2020-6(Type B) 
and assign pin numbers 3 & 6 to the EPs
and contribute it with the corresponding symbols of the 4 dual p-Channel Mosfets from Diodes?

evanshultz, 2019-03-01:
It looks like many, many companies use a largely-compatible package under a variety of names. See:
- https://www.infineon.com/dgdl/irlhs6376pbf.pdf?fileId=5546d462533600a401535663bef425af
- https://www.onsemi.com/pub/Collateral/FDMA2002NZ-D.pdf
- https://assets.nexperia.com/documents/data-sheet/PMDPB56XNEA.pdf
- http://www.mccsemi.com/up_pdf/MCMNP517(DFN2020-6U)-A.pdf
- http://aosmd.com/res/data_sheets/AON2810.pdf

 `DFN2020-6` or some similar variant is common, but so is just a riff on `DFN-6...`. Some call it `MicroFET...` and Infineon, for reasons that don't make sense to me since the package has two rows of pins, calls it a `PQFN`.

So let's go generic. But because the dimensions can vary we have a choice to make. As our scripted generator doesn't support this type of package, if we don't try a "one size fits all" approach we could end up with a lot of similar footprints that could support all packages. (Unique packages are best, but for this general library let's go generic for handmade footprints.) How about `DFN-6-2EP_2x2mm_P0.65mm_EP1.0x0.7mm`?

pboe, 2019-03-02:
OK let's go the generic way.

> DFN-6-2EP_2x2mm_P0.65mm_EP1.0x0.7mm
0.7mm for the EP seems to be more generic, i'll extend the size of the EPs.

How to deal with the EPs? Assign pin number 7 & 8 and what's the best way to connect them? Just connect them on the pcb? This will probably result in erc errors, or make a symbol with the pin 7 and 8? Is there any way to hide those 2 pins  in the symbol view and connect them internaly?

evanshultz, 2019-03-04:
There are two ways:
1. Assign the same pin number to the pads on the PCB. They will automatically connect.
1. Stack two pins with different numbers in the symbol and use unique footprint pin numbers.

Both ways work, but the former is simpler. So... if all of the parts using this footprint use the same EP connection (the few I checked all do) then we can use that method. Does that make sense and sound OK to you?

pboe, 2019-03-05:
Sounds very good for me. Started a pull request yesterday.

evanshultz, 2019-04-01:
Thanks. Will close this then.

BTW, I just noted that the On Semi datasheet above references JEDEC MO-229. The others have no industry-standard package reference or specifically have a space for one but it is blank (assume it's intentionally a manufacturer-specific package or the datasheet is older than the JEDEC spec).

closed 2019-04-01 by evanshultz

evanshultz, 2019-09-26:
To close an open issue here... the ON Semi datasheet mentions JEDEC MO-229 but MO-229 does not shown any packages with two EPs and the table mentioning the VCCC variation does in fact only show a single EP dimension. So ON Semi has made an error somewhere.