KiCad/kicad-footprints #1463: Missing TI DGN (S-PDSO-G8) or HVSSOP

rs-development, 2019-03-05:
I just created a new Symbol for the TPS7A16 Regulator but could not find the matching footprint. 

[http://www.ti.com/lit/ds/symlink/tps7a16.pdf](http://www.ti.com/lit/ds/symlink/tps7a16.pdf)

http://www.ti.com/lit/ml/mpds046e/mpds046e.pdf

Is the footprint existing somewhere or should i create it?

evanshultz, 2019-03-05:
There are a few footprints in Package_SO that could work. The body size and lead pitch are common, and we are given a range of thermal pad sizes in the datasheet (1.34-1.6x1.66-1.92mm).

To show footprints having a thermal pad and matching the pitch and body dimensions and an EP less than 2mm in he X direction, we can enter the following in the footprint editor: `8-1EP_3x3mm*P0.65mm_EP1`. I am using a recent nightly and see this:
[image]

I would be inclined to go with `MSOP-8-1EP_3x3mm_P0.65mm_EP1.73x1.85mm` since it's a new, scripted footprint and has nearly correct thermal pad dimensions. That should work fine if you want to use an existing footprint in your own designs.

However, it would be best to make a unique footprint just for this part so the thermal pad size matches the datasheet. I would like to see this done if you're submitting the symbol. With the footprint script it's quite easy to make a new footprint so I would create a new one if you want to submit this symbol, and its footprint, to the library.

Thanks!

rs-development, 2019-03-05:
Thanks for that 👍 

I will make the PR, if the part is used in our design.

rs-development, 2019-03-05:
Can you recommend the dimensions for the creation of the footprint. I can not find a manufacturer recommendation?

Pad Width in the middle from min and max?
Pad length is not well defined in datasheeet

Should the pad extent outside of the width of the package? From KLC i would guess it should not.

evanshultz, 2019-03-05:
An EP the maximum of the package's thermal pad should work. You can overwrite the EP size in the script if it uses the mean value.

Let's just keep the EP under the part body as the VSON handily wins with regards to thermals. I don't see much point in going crazy to add copper to the SSOP footprint if choosing the SON package will be dramatically better.

It looks like pad length is 0.4-0.7mm from page 31 of the datasheet. Are we talking about the same thing?

rs-development, 2019-03-05:
I was talking about Pins 1 to 8, i think the EP is quite clear.

I got confused with size restrictions of EP and normal pads. Are there any guidelines in the KLC regarding standard pad sizes?

evanshultz, 2019-03-05:
Our script handles this for you. Enter the package info and it will spit out the footprint. Are you using the script?

rs-development, 2019-03-05:
With script you refer to the wizard? The wizard seems to use pad values not component dimensions

[image]

evanshultz, 2019-03-06:
No, that thing is crap. I'm talking about https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Packages/Package_Gullwing__QFP_SOIC_SO.