KiCad/kicad-footprints #1487: Request for SMD 2.54mm Pin headers in Horizontal config

gcormier, 2019-03-19:
The library is full of every other type it would seem, minus the horizontal SMD configuration.

[image]

Trying to get python scripting going has been failure for me. Would anyone consider churning out a bunch of these?

TSM-106-01-T-SH
[image]

evanshultz, 2019-03-21:
We don't take requests as a rule since that would overwhelm everybody and since our footprint tools are freely-available, but we can help you with scripting. Submit a PR or issue at https://github.com/pointhi/kicad-footprint-generator, tag me if you like, and we'll see what we can do to get you going. It's fun (a nerdy kind of fun) once you're over the hump!

I'm not sure if this is generic and making them manufacturer-specific was wrong, but did you see:
[image]

gcormier, 2019-03-21:
Will do!

closed 2019-09-30 by gcormier