KiCad/kicad-footprints #1496: RTC footprint

jneiva08, 2019-03-22:
Hi,

Where is the best localization to put the footprint for [RV-1805-C3](https://www.microcrystal.com/fileadmin/Media/Products/RTC/Datasheet/RV-1805-C3.pdf)?

Have already made the symbol. When finished the footprint will open PR.

Thanks

jneiva08, 2019-03-23:
If put F.Silk 0.12 from F.Fab it will be less than 0.1 from PAD.
In this case can move it to 0.2 from PAD edge?

jneiva08, 2019-03-24:
Where is the proper place to put add this footprint?
Is 10 pins with 0.8 of pitch.

[image]

poeschlr, 2019-03-24:
Maybe the oscillator lib?

poeschlr, 2019-03-24:
Or a new RTC lib (named similarly to the respective symbol library.)

jneiva08, 2019-03-24:
Do you mean rtc.pretty?

jneiva08, 2019-03-27:
Have created the PR for this footprint, can be found  in https://github.com/KiCad/kicad-footprints/pull/1505