KiCad/kicad-footprints #1497: Keystone 2468 screw holes are backwards

ianrrees, 2019-03-25:
The footprint has two mounting holes, currently they are in the wrong position.  They are oriented as for mounting the battery pack on the back side of the board, where the rest of the footpring puts the battery pack on the front side.  See http://www.keyelco.com/product-pdf.cfm?p=1033

evanshultz, 2019-03-25:
Thanks for taking time to report this.

Datasheet:
[image]

Footprint:
[image]

It looks OK to me, but you have to rotate and orient things correctly since the footprint has the `+` pin at the top left. Can you please clarify?

ianrrees, 2019-03-25:
Basically, the Y (or X) coordinates of the two mounting holes need to be swapped with each other.

[image]

evanshultz, 2019-03-25:
Oh. I got it. I didn't catch that this was about the mounting holes. Sorry.

Are you willing to submit a PR to fix this issue?

ianrrees, 2019-03-25:
I'll make one based on the web client shortly, but didn't do that initially because the last time I've made PRs to KiCad repos, they seem to have gotten caught in a morass of editing and commentary that left me feeling like it was a waste of effort.

PR #1501 (2019-03-25): Fix #1497 mounting hole positions in Keystone 2468
Fix #1497 mounting hole positions in Keystone 2468

The issue has a photo of the part in question, and a PCB made with the original footprint.

http://www.keyelco.com/product-pdf.cfm?p=1033

---

All contributions to the kicad library must follow the [KiCad library convention](http://kicad-pcb.org/libraries/klc/)

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [x] Provide a URL to a datasheet for the footprint(s) you are contributing
- [ ] An exa [... cut]
commit eaef2a0f (2019-03-25): Fix #1497 mounting hole positions in Keystone 2468 (#1501)
closed 2019-03-25 by ianrrees