KiCad/kicad-footprints #1499: G6S-2 has pad numbers inconsistent with symbol

sxpert, 2019-03-25:
symbol shows correct pin assignments 8, 9, 10 and 12, according to the datasheet.
footprint has 6, 7, 8 and 10

evanshultz, 2019-03-25:
Thanks for noticing and reporting this issue.

Symbols:
[image]

Footprints:
[image]

Confirmed. Are you willing to submit a PR to fix the footprints?

Note that there are G6SK* footprints but no corresponding symbols. I've submitted it at https://github.com/KiCad/kicad-symbols/pull/1679.

gkeeth, 2019-05-27:
I believe the footprint numbering issue was fixed by #1563 (although I was not aware of this issue when I submitted the pull request!)

evanshultz, 2019-05-28:
@sxpert 
Please confirm if this resolves your issue. You may need to update your libraries to see the fix. I'll close this in a couple days if I don't see a response.