KiCad/kicad-footprints #1518: Incorrect footprint Molex micro-fit 43650-XX15

anymouse404, 2019-04-01:
Affected footprints are Molex_Micro-Fit_3.0_43650-??15_1x04_P3.00mm_Vertical where ?? is two-digit number of pins.
Two mechanical mounting holes in the corners have size 1.0 mm instead of 1.27 as recommended in datasheet.
Link to device datasheet is https://www.molex.com/pdm_docs/sd/436500415_sd.pdf
Attached is screenshot of datasheet combined with screenshot of footprint editor showing drill size.
[image]

Kicad version details are following:

Application: kicad
Version: 5.1.0-060a0da~80~ubuntu18.10.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.61.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.4) nghttp2/1.32.1 librtmp/2.3
Platform: Linux 4.18.0-16-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.61.0
    Compiler: GCC 8.2.0 with C++ ABI 1013

evanshultz, 2019-04-01:
Confirmed. Thanks for reporting this and sorry for the error in our library.

None of these are using generic scripts, so I checked all Micro-Fit 3.0 scripts. Issues found:
- https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_micro-fit-3.0_tht_top_single_row.py, with the error reported above on line 55. Also, line 54 should be 1.02mm and not 1mm (small, but not to the doc.
- https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_micro-fit-3.0_tht_top_dual_row.py, lines 54 and 55 should be 1.02mm (see https://www.molex.com/pdm_docs/sd/430450212_sd.pdf)

Good scripts (at least with regards to pin and peg drill hole diameters):
- https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_micro-fit-3.0_tht_side_single_row.py
- https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_micro-fit-3.0_tht_side_dual_row.py
- https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_micro-fit-3.0_smd_top_dual_row.py
- https://github.com/pointhi/kicad-footprint-generator/blob/master/scripts/Connector/Connector_Molex/conn_molex_micro-fit-3.0_smd_side_dual_row.py (all SMD)

I'll update the scripts and push corrected footprints tomorrow unless somebody else gets to it.

evanshultz, 2019-04-02:
Thanks again for this report.

Script fix: https://github.com/pointhi/kicad-footprint-generator/pull/334
Footprint fix: https://github.com/KiCad/kicad-footprints/pull/1520

closed 2019-04-02 by poeschlr